bugsounet/MMM-GoogleAssistant #2: Problème Google Assistant

jeff44850, 2020-03-25:
Bonjour
Après l'installation et l'utilisation avec le mot clé  "jarvis" le module fonctionne correctement sur quelques essais ( nombre aléatoire ) et par la suite à la détection du mot clé Magic Mirror redémarre !
Avez-vous déjà entendu parler du problème ou une idée ?
cordialement

bugsounet, 2020-03-25:
Bonjour Jeff,

Oui, je confirme. Jj'ai remarquer cela aussi, par moment il ne veux pas détecter de mot clé... et tout a coup, quelques minute/ seconde plus tard cela remarche.

J'arrive a atténuer le problème en activant le mode `debug` dans le fichier de configuration.
Je vais essayer de tracer/pister le problème afin de corriger

bugsounet, 2020-04-20:
voila le problème est maintenant résolu.
Essayer de mettre a jours vers la version 1.1.0

Désolé pour le retard,
Cédric

jeff44850, 2020-04-20:
Merci beaucoup
Je met à jour très vite
Bonne journée
Cordialement

closed 2020-05-02 by bugsounet