bugsounet/MMM-GoogleAssistant #21: i`m a student who used hotwords ang

leejaehoon24, 2020-06-02:
while installing hotwords. there`s an error.

npm WARN deprecated request@2.88.2: request has been deprecated, see https://github.com/request/request/issues/3142
npm ERR! code E404
npm ERR! 404 Not Found: @bugsounet/lpcm16@^1.0.1

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2020-06-02T11_46_01_081Z-debug.log

so i deleted this sentence from the pakage.

pakage
"dependencies": {
    "@bugsounet/buffertowav": "^1.0.1",
    "@bugsounet/lpcm16": "^1.0.1",

after installation, we started npm, but the error occurred in node_help, so we deleted this sentence and  excuted it

const Record = require("@bugsounet/lpcm16")

npm start is possible but voice recognition is not possible.
what should we do?

bugsounet, 2020-06-02:
MMM-Hotword and MMM-AssistantMk2 project is now closed or deprecied
So we can't help you about this 2 projects

You can use this new project if you want

closed 2020-06-02 by bugsounet

leejaehoon24, 2020-06-02:
can this google assistant function ai function because it is voice recognition?

leejaehoon24, 2020-06-02:
Is voice recognition and ai supported only by Google Assistant emojis?