bugsounet/MMM-GoogleAssistant #22: Does Google Assistant support voice recognition and ai?

leejaehoon24, 2020-06-02:
if possible, I get this error, so could you help me?

pi@raspberrypi:~/MagicMirror $ npm start

> magicmirror@2.11.0 start /home/pi/MagicMirror
> DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js

[2020-06-02 21:42:29.696] [LOG]    Starting MagicMirror: v2.11.0
[2020-06-02 21:42:29.742] [LOG]    Loading config ...
[2020-06-02 21:42:29.750] [LOG]    Loading module helpers ...
[2020-06-02 21:42:29.753] [LOG]    No helper found for module: alert.
[2020-06-02 21:42:29.834] [LOG]    Initializing new module helper ...
[2020-06-02 21:42:29.835] [LOG]    Module helper loaded: updatenotification
[2020-06-02 21:42:29.836] [LOG]    No helper found for module: clock.
[2020-06-02 21:42:30.462] [LOG]    Initializing new module helper ...
[2020-06-02 21:42:30.463] [LOG]    Module helper loaded: calendar
[2020-06-02 21:42:30.464] [LOG]    No helper found for module: compliments.
[2020-06-02 21:42:30.465] [LOG]    No helper found for module: currentweather.
[2020-06-02 21:42:30.466] [LOG]    No helper found for module: weatherforecast.
[2020-06-02 21:42:30.506] [LOG]    Initializing new module helper ...
[2020-06-02 21:42:30.507] [LOG]    Module helper loaded: newsfeed
[2020-06-02 21:42:33.095] [LOG]    Initializing new module helper ...
[2020-06-02 21:42:33.096] [LOG]    Module helper loaded: MMM-GoogleAssistant
[2020-06-02 21:42:33.097] [LOG]    All module helpers loaded.
[2020-06-02 21:42:33.373] [LOG]    Starting server on port 8080 ... 
[2020-06-02 21:42:33.401] [LOG]    Server started ...
[2020-06-02 21:42:33.402] [LOG]    Connecting socket for: updatenotification
[2020-06-02 21:42:33.404] [LOG]    Connecting socket for: calendar
[2020-06-02 21:42:33.405] [LOG]    Starting node helper for: calendar
[2020-06-02 21:42:33.405] [LOG]    Connecting socket for: newsfeed
[2020-06-02 21:42:33.406] [LOG]    Starting module: newsfeed
[2020-06-02 21:42:33.407] [LOG]    Connecting socket for: MMM-GoogleAssistant
[2020-06-02 21:42:33.408] [LOG]    Sockets connected & modules started ...
[2020-06-02 21:42:34.003] [LOG]    Launching application.
[2020-06-02 21:42:38.543] [LOG]    Create new calendar fetcher for url: http://www.calendarlabs.com/ical-calendar/ics/76/US_Holidays.ics - Interval: 300000
[2020-06-02 21:42:38.616] [LOG]    Create new news fetcher for url: http://www.nytimes.com/services/xml/rss/nyt/HomePage.xml - Interval: 300000
[2020-06-02 21:42:38.754] [LOG]    [ASSISTANT] MMM-GoogleAssistant Version: 2.0.0
[2020-06-02 21:42:38.757] [LOG]    [SNOWBOY] Model selected: jarvis
[2020-06-02 21:42:38.792] [LOG]    [SNOWBOY] snowboy v1.5.0 Initialized...
[2020-06-02 21:42:38.796] [LOG]    [ASSISTANT] Google Assistant is initialized.
[2020-06-02 21:42:39.136] [LOG]    [LPCM16] WARN: arecord: main:828: audio open error: 그런 파일이나 디렉터리가 없습니다

[2020-06-02 21:42:39.141] [LOG]    [SNOWBOY] Retry restarting...
[2020-06-02 21:42:41.605] [LOG]    [LPCM16] WARN: arecord: main:828: audio open error: 그런 파일이나 디렉터리가 없습니다

[2020-06-02 21:42:41.608] [LOG]    [SNOWBOY] Retry restarting...
[2020-06-02 21:42:43.688] [LOG]    [LPCM16] WARN: arecord: main:828: 
[2020-06-02 21:42:43.690] [LOG]    [LPCM16] WARN: audio open error: 그런 파일이나 디렉터리가 없습니다

[2020-06-02 21:42:43.694] [LOG]    [SNOWBOY] Retry restarting...
[2020-06-02 21:42:45.793] [LOG]    [LPCM16] WARN: arecord: main:828: 
[2020-06-02 21:42:45.794] [LOG]    [LPCM16] WARN: audio open error: 그런 파일이나 디렉터리가 없습니다

[2020-06-02 21:42:45.796] [LOG]    [SNOWBOY] Retry restarting...
[2020-06-02 21:42:47.899] [LOG]    [LPCM16] WARN: arecord: main:828: 
[2020-06-02 21:42:47.901] [LOG]    [LPCM16] WARN: audio open error: 그런 파일이나 디렉터리가 없습니다

[2020-06-02 21:42:47.903] [LOG]    [SNOWBOY] Retry restarting...
[2020-06-02 21:42:50.021] [LOG]    [LPCM16] WARN: arecord: main:828: 
[2020-06-02 21:42:50.023] [LOG]    [LPCM16] WARN: audio open error: 그런 파일이나 디렉터리가 없습니다

[2020-06-02 21:42:50.025] [LOG]    [SNOWBOY] Retry restarting...
[2020-06-02 21:42:52.101] [LOG]    [LPCM16] WARN: arecord: main:828: 
[2020-06-02 21:42:52.102] [LOG]    [LPCM16] WARN: audio open error: 그런 파일이나 디렉터리가 없습니다

[2020-06-02 21:42:52.104] [LOG]    [SNOWBOY] Retry restarting...
[2020-06-02 21:42:54.166] [LOG]    [LPCM16] WARN: arecord: main:828: 
[2020-06-02 21:42:54.168] [LOG]    [LPCM16] WARN: audio open error: 그런 파일이나 디렉터리가 없습니다

[2020-06-02 21:42:54.169] [LOG]    [SNOWBOY] Retry restarting...
[2020-06-02 21:42:56.265] [LOG]    [LPCM16] WARN: arecord: main:828: 
[2020-06-02 21:42:56.266] [LOG]    [LPCM16] WARN: audio open error: 그런 파일이나 디렉터리가 없습니다

[2020-06-02 21:42:56.268] [LOG]    [SNOWBOY] Retry restarting...
[2020-06-02 21:42:58.374] [LOG]    [LPCM16] WARN: arecord: main:828: 
[2020-06-02 21:42:58.376] [LOG]    [LPCM16] WARN: audio open error: 그런 파일이나 디렉터리가 없습니다

[2020-06-02 21:42:58.378] [LOG]    [SNOWBOY] Retry restarting...
[2020-06-02 21:43:00.464] [LOG]    [LPCM16] WARN: arecord: main:828: 
[2020-06-02 21:43:00.466] [LOG]    [LPCM16] WARN: audio open error: 그런 파일이나 디렉터리가 없습니다

[2020-06-02 21:43:00.468] [LOG]    [SNOWBOY] Retry restarting...
[2020-06-02 21:43:02.543] [LOG]    [LPCM16] WARN: arecord: main:828: 
[2020-06-02 21:43:02.546] [LOG]    [LPCM16] WARN: audio open error: 그런 파일이나 디렉터리가 없습니다

[2020-06-02 21:43:02.551] [LOG]    [SNOWBOY] Retry restarting...
[2020-06-02 21:43:04.653] [LOG]    [LPCM16] WARN: arecord: main:828: 
[2020-06-02 21:43:04.655] [LOG]    [LPCM16] WARN: audio open error: 그런 파일이나 디렉터리가 없습니다

[2020-06-02 21:43:04.657] [LOG]    [SNOWBOY] Retry restarting...
[2020-06-02 21:43:06.748] [LOG]    [LPCM16] WARN: arecord: main:828: 
[2020-06-02 21:43:06.750] [LOG]    [LPCM16] WARN: audio open error: 그런 파일이나 디렉터리가 없습니다

[2020-06-02 21:43:06.751] [LOG]    [SNOWBOY] Retry restarting...
[2020-06-02 21:43:08.817] [LOG]    [LPCM16] WARN: arecord: main:828: 
[2020-06-02 21:43:08.819] [LOG]    [LPCM16] WARN: audio open error: 그런 파일이나 디렉터리가 없습니다

[2020-06-02 21:43:08.823] [LOG]    [SNOWBOY] Retry restarting...
[2020-06-02 21:43:10.902] [LOG]    [LPCM16] WARN: arecord: main:828: 
[2020-06-02 21:43:10.904] [LOG]    [LPCM16] WARN: audio open error: 그런 파일이나 디렉터리가 없습니다

[2020-06-02 21:43:10.905] [LOG]    [SNOWBOY] Retry restarting...
[2020-06-02 21:43:12.999] [LOG]    [LPCM16] WARN: arecord: main:828: 
[2020-06-02 21:43:13.001] [LOG]    [LPCM16] WARN: audio open error: 그런 파일이나 디렉터리가 없습니다

[2020-06-02 21:43:13.003] [LOG]    [SNOWBOY] Retry restarting...
[2020-06-02 21:43:15.111] [LOG]    [LPCM16] WARN: arecord: main:828: 
[2020-06-02 21:43:15.114] [LOG]    [LPCM16] WARN: audio open error: 그런 파일이나 디렉터리가 없습니다

[2020-06-02 21:43:15.117] [LOG]    [SNOWBOY] Retry restarting...
[2020-06-02 21:43:17.223] [LOG]    [LPCM16] WARN: arecord: main:828: 
[2020-06-02 21:43:17.225] [LOG]    [LPCM16] WARN: audio open error: 그런 파일이나 디렉터리가 없습니다

[2020-06-02 21:43:17.229] [LOG]    [SNOWBOY] Retry restarting...
[2020-06-02 21:43:19.336] [LOG]    [LPCM16] WARN: arecord: main:828: 
[2020-06-02 21:43:19.338] [LOG]    [LPCM16] WARN: audio open error: 그런 파일이나 디렉터리가 없습니다

[2020-06-02 21:43:19.341] [LOG]    [SNOWBOY] Retry restarting...
[2020-06-02 21:43:19.713] [LOG]    Shutting down server...
[2020-06-02 21:43:19.714] [LOG]    Stopping module helper: updatenotification
[2020-06-02 21:43:19.715] [LOG]    Stopping module helper: calendar
[2020-06-02 21:43:19.716] [LOG]    Stopping module helper: newsfeed
[2020-06-02 21:43:19.716] [LOG]    Stopping module helper: MMM-GoogleAssistant

bugsounet, 2020-06-02:
It's means... your micConfig device is not correct

leejaehoon24, 2020-06-02:
here my micConfig 
what`s problem?
		{
  module: "MMM-GoogleAssistant",
  position: "fullscreen_above",
  config: {
    assistantConfig: {
      latitude: 51.508530,
      longitude: -0.076132,
    },
    micConfig: { // put there configuration generated by auto-installer
      recorder: "arecord",
      device: "plughw:1",
    },
  }
},

bugsounet, 2020-06-02:
verify your device card with `arecord -l`

leejaehoon24, 2020-06-02:
this is it
pi@raspberrypi:~/MagicMirror/modules $ arecord -l
**** List of CAPTURE Hardware Devices ****
card 2: Device [USB PnP Sound Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

bugsounet, 2020-06-02:
so ... you can see it's the card 2.
in your config you have selected the card 1.
just change it ;)

leejaehoon24, 2020-06-02:
thank you . thank you for the correction. there was a file called "with-Youtube.js" in the ricipe before,
but it`s not here. Could you write it for me?

bugsounet, 2020-06-02:
GA is not compatible with MMM-Youtube
you can use [MMM-Assistant2Display](https://github.com/bugsounet/MMM-Assistant2Display) for YouTube functionality