bugsounet/MMM-GoogleAssistant #28: 2 issues - maybe related

johnpaularthur, 2020-06-07:
Hi, I'm trying to get this running.  It is a new install of Raspberry OS on a PI 3 A+.   I have magicmirror working with two additional modules installed; MMM-Spotify and MMM-Snowboy (however, the later is not being used yet that I know of).

The first issue is that during `npm install` of MMM-GoogleAssistant, the speaker and mic test failed with "current configuration not working !".  If i skip the test, the install completes.   Even tho the test fail, I can play raspotify and youtube thru the speakers just fine.

Second issue which might be related is once I enable the module in magicmirror config, I get the following error below and MagicMirror fails to load.   Have I missed a step somewhere?

```[2020-06-07 03:00:06.197] [ERROR]  WARNING! Could not load config file. Starting with default configuration. Error found: Error: Cannot find module './lib/node/index.js
'
Require stack:
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/index.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_helper.js
- /home/pi/MagicMirror/js/app.js
- /home/pi/MagicMirror/js/electron.js
- /home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js
- 
[2020-06-07 03:00:06.205] [ERROR]  App threw an error during load
[2020-06-07 03:00:06.207] [ERROR]  Error: Cannot find module './lib/node/index.js'
Require stack:
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/index.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_helper.js
- /home/pi/MagicMirror/js/app.js
- /home/pi/MagicMirror/js/electron.js
- /home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js
- 
    at Module._resolveFilename (internal/modules/cjs/loader.js:627:15)
    at Function.Module._resolveFilename (/home/pi/MagicMirror/node_modules/electron/dist/resources/electron.asar/common/reset-search-paths.js:41:16)
    at Function.Module._resolveFilename (/home/pi/MagicMirror/node_modules/module-alias/index.js:49:29)
    at Function.Module._load (internal/modules/cjs/loader.js:531:27)
    at Module.require (internal/modules/cjs/loader.js:685:19)
    at require (internal/modules/cjs/helpers.js:16:16)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/index.js:6:18)
    at Module._compile (internal/modules/cjs/loader.js:786:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:798:10)
    at Module.load (internal/modules/cjs/loader.js:645:32)
[2020-06-07 03:00:06.210] [ERROR]  Error: Cannot find module './lib/node/index.js'
Require stack:
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/index.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_helper.js
- /home/pi/MagicMirror/js/app.js
- /home/pi/MagicMirror/js/electron.js
- /home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js
- 
    at Module._resolveFilename (internal/modules/cjs/loader.js:627:15)
    at Function.Module._resolveFilename (/home/pi/MagicMirror/node_modules/electron/dist/resources/electron.asar/common/reset-search-paths.js:41:16)
    at Function.Module._resolveFilename (/home/pi/MagicMirror/node_modules/module-alias/index.js:49:29)
    at Function.Module._load (internal/modules/cjs/loader.js:531:27)
    at Module.require (internal/modules/cjs/loader.js:685:19)
    at require (internal/modules/cjs/helpers.js:16:16)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/index.js:6:18)
    at Module._compile (internal/modules/cjs/loader.js:786:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:798:10)
    at Module.load (internal/modules/cjs/loader.js:645:32) {
  code: 'MODULE_NOT_FOUND',
  requireStack: [
    '/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/index.js',
    '/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_helper.js',
    '/home/pi/MagicMirror/js/app.js',
    '/home/pi/MagicMirror/js/electron.js',
    '/home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js',
    undefined
  ]
}
```

bugsounet, 2020-06-07:
1) mic test will be updated soon

2) snowboy library is not installed (or electron rebuild part).
retry complete installation

johnpaularthur, 2020-06-07:
Awesome.  A full delete and rebuild worked.  Thanks!

closed 2020-06-07 by johnpaularthur