bugsounet/MMM-GoogleAssistant #32: Voice activation issues on MMM-GoogleAssistant

clify007, 2020-06-08:
Hello,

I am facing issues with voice activation on the assistant when my MM is running.

I am able to /query the assistant through telegram and everything works, I can use voice command normally and get a valid response through the speakers. I need help on how to keep the assistant listening for the hotword without having to use telegram app everytime.

On a side note, having been trying to install MMM-Hotword but somehow  I am ending up with an error showing repository not found when I try git clone.
For clarity, I am very much a newbie and  would appreciate all the help I can get.

Modules running on my magic mirror : MMM-RemoteControl, MMM-GoogleAssistant, MMM-Telegram, MMM-Internetmonitor, MMM-Instagram, MMM-Covid19, MMM-Selfieshot, MMM-NowPlayingOnSpotify in addition to default modules.

MM working normally excerpt for the voice activation part.

Thanks in advance.

bugsounet, 2020-06-09:
i have updated checkmic part.

use `npm run update` to update

and it will help you to determinate your micConfig {}

clify007, 2020-06-17:
Hi,
I have run the update and things are working now. I had to reposition the mic as well.
No issues so far. Moving on to custom commands.

Thanks for all the help.

closed 2020-06-18 by bugsounet