bugsounet/MMM-GoogleAssistant #36: responseConfig issue

chanapai, 2020-06-11:
Hi, 
I have a trouble after finished install. 
When I call "Jarvis", the white bar appear on the top of the monitor but not thing happen next. It just still not hide, no sound and any text.

my config is simple 
`{
                        module: "MMM-GoogleAssistant",
                        position: "fullscreen_above",
                        config: {
                                assistantConfig: {
                                        latitude: 51.508530,
                                        longitude: -0.076132,
                                },
                               micConfig: { // put there configuration generated by auto-installer
                                        recorder: "arecord",
                                        device: "plughw:1",
                                },
                                snowboy: {
                                        audioGain: 2.0,
                                        Frontend: true,
                                        Model: "jarvis",
                                        Sensitivity: null
                                },
                      }
},`

Thank in advance.

PineMaster, 2020-06-11:
I mean you gotta ask it a question or smth ;)

chanapai, 2020-06-11:
I have tried to install new one and it work fine.
may be I have installed many module or didn't update something.

closed 2020-06-11 by chanapai