bugsounet/MMM-GoogleAssistant #37: Change audio source

hm1234, 2020-06-11:
I am trying to use my google aiy hat speaker and mic. I got it working mic works and trigger but the speaker I only hear the chime but does not play any audio. I wanted to know if there is somewhere to  change the audio source?

bugsounet, 2020-06-11:
you can audio change source is on the desktop (volume icon)
MMM-GoogleAssistant is default output with HTML5

after if you only hear the chime try this:

```sh
apt list --installed | grep pulseaudio
```

if installed ... This module is not compatible with this one
I have actually no solutions for use it

hm1234, 2020-06-11:
Thank you I'll try it

hm1234, 2020-06-11:
@bugsounet I got things working just a quick question while the assistant talking is there a way to trigger it while talking like stop it. I see that the mic turns off while the assistant is talking and turn back on when it's done?

bugsounet, 2020-06-12:
unfortunately, this is not planned

closed 2020-06-13 by hm1234