bugsounet/MMM-GoogleAssistant #40: Command gpio pins

0rion12, 2020-06-12:
Hello,
I would like to know if it is possible to drive the raspberry gpio using a predefined command. For example, if you say "Jarvis... Switch on the light" it should activate the gpio 4 (where a led is plugged) and so switch on the led.

bugsounet, 2020-06-15:
Hi, you can make your own recipe with a transcriptionHook and shell exec for execute shell command.
You have an exemple with [this recipe](https://github.com/bugsounet/MMM-GoogleAssistant/blob/prod/recipes/Reboot-Restart-Shutdown.js)

juan-np, 2020-06-22:
Hi, @bugsounet , thank so much for your great work. I did exactly that and now is working nice. I used the .sh sample of my PiRelay board for command on the Shell Exec part.

closed 2020-06-23 by bugsounet