bugsounet/MMM-GoogleAssistant #41: Black Screen

Joshsamuel101, 2020-06-13:


Joshsamuel101, 2020-06-13:
> /* Magic Mirror Config Sample
 *
 * By Michael Teeuw http://michaelteeuw.nl
 * MIT Licensed.
 *
 * For more information on how you can configure this file
 * See https://github.com/MichMich/MagicMirror#configuration
 *
 */

var config = {
	address: "localhost", // Address to listen on, can be:
	                      // - "localhost", "127.0.0.1", "::1" to listen on loopback interface
	                      // - another specific IPv4/6 to listen on a specific interface
	                      // - "0.0.0.0", "::" to listen on any interface
	                      // Default, when address config is left out or empty, is "localhost"
	port: 8080,
	ipWhitelist: ["127.0.0.1", "::ffff:127.0.0.1", "::1"], // Set [] to allow all IP addresses
	                                                       // or add a specific IPv4 of 192.168.1.5 :
	                                                       // ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.1.5"],
	                                                       // or IPv4 range of 192.168.3.0 --> 192.168.3.15 use CIDR format :
	                                                       // ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.3.0/28"],

	useHttps: false, 		// Support HTTPS or not, default "false" will use HTTP
	httpsPrivateKey: "", 	// HTTPS private key path, only require when useHttps is true
	httpsCertificate: "", 	// HTTPS Certificate path, only require when useHttps is true

	language: "en",
	timeFormat: 12,
	units: "metric",
	// serverOnly:  true/false/"local" ,
			     // local for armv6l processors, default
			     //   starts serveronly and then starts chrome browser
			     // false, default for all  NON-armv6l devices
			     // true, force serveronly mode, because you want to.. no UI on this device

	modules: [
		{
			module: "alert",
		},
		{
			module: "updatenotification",
			position: "top_bar"
		},
		{
			module: "clock",
			position: "top_left"
		},
		{
			module: "calendar",
			header: "My Calendar",
			position: "top_left",
			config: {
				calendars: [
					{
						symbol: "calendar-check-o ",
						url: "webcal://www.calendarlabs.com/ical-calendar/ics/75/UK_Holidays.ics"  
          		                },
			                {	symbol: "MyCalendar",
                                                url: "https://calendar.google.com/calendar/ical/joshysamuel101%40gmail.com/private-b11566dd084bd1d1277ded071f1b7c4f/basic.ics"                               
			                }
	
                                ]         	
                         }
                },
		{
			module: "currentweather",
			position: "top_right",
			config: {
				location: "London",
[... 66 lines cut ...]
                          },
                          customActionConfig: {
                              autoMakeAction: false,
                              autoUpdateAction: false, // in RPI, gaction CLI might have some trouble.(current version should be 2.2.4, but for linux-arm, Google haven't updated) so leave this as false in RPI. I don't know it is solved or not.
                              actionLocale: "en-US", // At this moment, multi-languages are not supported, sorry. Someday I'll work.
                         },
                         snowboy: {
                              audioGain: 2.0,
                              Frontend: true,
                              Model: "jarvis",
                              Sensitivity: null
                         },
                         A2DServer: {
                              useA2D: false,
                              stopCommand: "stop"
                         },
                         recipes: [ "with-MMM-TelegramBot.js" ]
                       }
              },
       ]

};

/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== "undefined") {module.exports = `config;}`

Joshsamuel101, 2020-06-13:
0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start', 'dev' ]
2 info using npm@6.14.4
3 info using node@v10.21.0
4 verbose config Skipping project config: /home/pi/.npmrc. (matches userconfig)
5 verbose stack Error: ENOENT: no such file or directory, open '/home/pi/package.json'
6 verbose cwd /home/pi
7 verbose Linux 4.19.118-v7l+
8 verbose argv "/usr/bin/node" "/usr/bin/npm" "start" "dev"
9 verbose node v10.21.0
10 verbose npm  v6.14.4
11 error code ENOENT
12 error syscall open
13 error path /home/pi/package.json
14 error errno -2
15 error enoent ENOENT: no such file or directory, open '/home/pi/package.json'
16 error enoent This is related to npm not being able to find a file.
17 verbose exit [ -2, true ]

bugsounet, 2020-06-14:
what is this log ?

Joshsamuel101, 2020-06-14:
> what is this log ?

`npm start dev`

Joshsamuel101, 2020-06-14:
Hi @bugsounet, I've tried doing this again and I remember I was experiencing the following issue after inputting the following code:

`curl -s -X POST -H "Content-Type: application/json" \
-H "Authorization: Bearer $ACCESSTOKEN" -d @deviceInstance.json \
https://embeddedassistant.googleapis.com/v1alpha2/projects/YOUR_PROJECT_ID/devices/`

My result was:
`{
  "error": {
    "code": 404,
    "message": "Requested entity was not found.",
    "status": "NOT_FOUND"
  }
}`

Not too sure where to go from here, let me know how I can fix this thanks

lxne, 2020-06-14:
After a fresh installation did I get a black screen, too.
I went into the MMM-GoogleAssistant folder and executed `npm run update`.
I chose not to rebuild electron but the script did it nonetheless.
Afterwards everything was working magically.

Joshsamuel101, 2020-06-14:
> After a fresh installation I did get a black screen, too.
> I went into the MMM-GoogleAssistant folder and executed `npm run update`.
> I chose not to rebuild electron but the script did it nonetheless.
> Afterwards everything was working magically.
> 
> PS: I do not know if it matters but, when I installed the assistant I did `npm audit fix` because of 6 found vulnerabilities. But when I did the update command I didn't do the audit fix.

Amazingly enough, I did exactly what you said (including the script forcefully rebuilding electron) and no black screen! Although I haven't got a microphone that is compatible yet, I should be receiving one tomorrow so I will be able to fully test then. Thank you so much!

Is there any ideas about my above error when registering the mirror as a google assistant device?
`{ "error": { "code": 404, "message": "Requested entity was not found.", "status": "NOT_FOUND" } }`

bugsounet, 2020-06-15:
> curl -s -X POST -H "Content-Type: application/json" \ -H "Authorization: Bearer $ACCESSTOKEN" -d @deviceInstance.json \ https://embeddedassistant.googleapis.com/v1alpha2/projects/YOUR_PROJECT_ID/devices/

don't forget to replace **YOUR_PROJECT_ID** value ;)

Joshsamuel101, 2020-06-15:
> > curl -s -X POST -H "Content-Type: application/json" \ -H "Authorization: Bearer $ACCESSTOKEN" -d @deviceInstance.json \ https://embeddedassistant.googleapis.com/v1alpha2/projects/YOUR_PROJECT_ID/devices/
> 
> don't forget to replace **YOUR_PROJECT_ID** value ;)

I entered my project ID and it still hasn't worked after multiple attempts :/

Kentipi, 2020-06-16:
Npm run update -> Electron rebuild? -> Yes. Fixed it for me

bugsounet, 2020-06-16:
hi, All step is needed :)

Joshsamuel101, 2020-06-16:
Last question from me, really appreciate all the help and support.

Is there a way to allow playback via HDMI (TV), and capture via USB? At the moment it's forcing me to select 1 sound card to run both

bugsounet, 2020-06-16:
In GA, this no configuration on this module for output device.
If you want to change output device, you need to set it in the RPI desktop : volume icon and set  your output device by default
Only micConfig is needed

bugsounet, 2020-06-23:
main issue is now solved.

closed 2020-06-23 by bugsounet