bugsounet/MMM-GoogleAssistant #42: The strange error during installation.

kffiatek, 2020-06-14:
Hey!
I tried everything but I can't make it through this point. This doesn't make sense to me, because the error is not even in the module, but somewhere in the node libraries. I provide the log from the moment when installation starts to brake:
 
`  CXX(target) Release/obj.target/grpc_node/ext/call.o
../ext/call.cc: In function ‘bool grpc::node::CreateMetadataArray(v8::Local<v8::Object>, grpc_metadata_array*)’:
../ext/call.cc:104:58: error: no matching function for call to ‘v8::Array::Get(unsigned int&)’
     Local<String> current_key(Nan::To<String>(keys->Get(i)).ToLocalChecked());
                                                          ^
In file included from /home/pi/.cache/node-gyp/14.4.0/include/node/node.h:67,
                 from ../ext/call.cc:23:
/home/pi/.cache/node-gyp/14.4.0/include/node/v8.h:3686:43: note: candidate: ‘v8::MaybeLocal<v8::Value> v8::Object::Get(v8::Local<v8::Context>, v8::Local<v8::Value>)’
   V8_WARN_UNUSED_RESULT MaybeLocal<Value> Get(Local<Context> context,
                                           ^~~
/home/pi/.cache/node-gyp/14.4.0/include/node/v8.h:3686:43: note:   candidate expects 2 arguments, 1 provided
/home/pi/.cache/node-gyp/14.4.0/include/node/v8.h:3689:43: note: candidate: ‘v8::MaybeLocal<v8::Value> v8::Object::Get(v8::Local<v8::Context>, uint32_t)’
   V8_WARN_UNUSED_RESULT MaybeLocal<Value> Get(Local<Context> context,
                                           ^~~
/home/pi/.cache/node-gyp/14.4.0/include/node/v8.h:3689:43: note:   candidate expects 2 arguments, 1 provided
../ext/call.cc: In static member function ‘static Nan::NAN_METHOD_RETURN_TYPE grpc::node::Call::StartBatch(Nan::NAN_METHOD_ARGS_TYPE)’:
../ext/call.cc:681:35: error: no matching function for call to ‘v8::Object::Get(uint32_t&)’
     if (!op->ParseOp(obj->Get(type), &ops[i])) {
                                   ^
In file included from /home/pi/.cache/node-gyp/14.4.0/include/node/node.h:67,
                 from ../ext/call.cc:23:
/home/pi/.cache/node-gyp/14.4.0/include/node/v8.h:3686:43: note: candidate: ‘v8::MaybeLocal<v8::Value> v8::Object::Get(v8::Local<v8::Context>, v8::Local<v8::Value>)’
   V8_WARN_UNUSED_RESULT MaybeLocal<Value> Get(Local<Context> context,
                                           ^~~
/home/pi/.cache/node-gyp/14.4.0/include/node/v8.h:3686:43: note:   candidate expects 2 arguments, 1 provided
/home/pi/.cache/node-gyp/14.4.0/include/node/v8.h:3689:43: note: candidate: ‘v8::MaybeLocal<v8::Value> v8::Object::Get(v8::Local<v8::Context>, uint32_t)’
   V8_WARN_UNUSED_RESULT MaybeLocal<Value> Get(Local<Context> context,
                                           ^~~
/home/pi/.cache/node-gyp/14.4.0/include/node/v8.h:3689:43: note:   candidate expects 2 arguments, 1 provided
make: *** [grpc_node.target.mk:202: Release/obj.target/grpc_node/ext/call.o] Error 1
make: Leaving directory '/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/grpc/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:194:23)
gyp ERR! stack     at ChildProcess.emit (events.js:315:20)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:276:12)
gyp ERR! System Linux 4.19.118-v7l+
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "build" "--fallback-to-build" "--library=static_library" "--module=/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/grpc/src/node/extension_binary/node-v83-linux-arm-glibc/grpc_node.node" "--module_name=grpc_node" "--module_path=/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/grpc/src/node/extension_binary/node-v83-linux-arm-glibc" "--napi_version=6" "--node_abi_napi=napi" "--napi_build_version=0" "--node_napi_label=node-v83"
gyp ERR! cwd /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/grpc
gyp ERR! node -v v14.4.0
gyp ERR! node-gyp -v v5.1.0
gyp ERR! not ok 
node-pre-gyp ERR! build error 
node-pre-gyp ERR! stack Error: Failed to execute '/usr/bin/node /usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --library=static_library --module=/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/grpc/src/node/extension_binary/node-v83-linux-arm-glibc/grpc_node.node --module_name=grpc_node --module_path=/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/grpc/src/node/extension_binary/node-v83-linux-arm-glibc --napi_version=6 --node_abi_napi=napi --napi_build_version=0 --node_napi_label=node-v83' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/grpc/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:315:20)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:1051:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:287:5)
node-pre-gyp ERR! System Linux 4.19.118-v7l+
node-pre-gyp ERR! command "/usr/bin/node" "/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/grpc/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build" "--library=static_library"
node-pre-gyp ERR! cwd /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/grpc
node-pre-gyp ERR! node -v v14.4.0
node-pre-gyp ERR! node-pre-gyp -v v0.13.0
node-pre-gyp ERR! not ok 
Failed to execute '/usr/bin/node /usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --library=static_library --module=/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/grpc/src/node/extension_binary/node-v83-linux-arm-glibc/grpc_node.node --module_name=grpc_node --module_path=/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/grpc/src/node/extension_binary/node-v83-linux-arm-glibc --napi_version=6 --node_abi_napi=napi --napi_build_version=0 --node_napi_label=node-v83' (1)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! grpc@1.23.4 install: `node-pre-gyp install --fallback-to-build --library=static_library`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the grpc@1.23.4 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2020-06-14T10_50_09_047Z-debug.log
`
I'm trying on Raspbian buster with all updater, Raspberry Pi 4. Please help.

PineMaster, 2020-06-15:
on what installation step did you get this error?

kffiatek, 2020-06-16:
As from my perspective installation has only one step, I have to provide the entire output. There is definitely something wrong with grpc.
[MMM-GoogleAssistant install log.txt](https://github.com/bugsounet/MMM-GoogleAssistant/files/4784489/MMM-GoogleAssistant.install.log.txt)

bugsounet, 2020-06-16:
hi, MagicMirror and GoogleAssistant (and google grpc library) use nodejs v10.x
actually, you use v14.x
downgrade it to v10 and it should works

kffiatek, 2020-06-16:
OK, that helped, thanks. I installed everything without errors, it runs smoothly, no console errors either. There is another problem thou. It does nothing. Does not give briefing on startup as Mk2 did, does not react on anything i tried "ok google", "hey google", "smart mirror", mirror mirror on the wall". Microphone and speaker work. Any clues? Maybe credentials taken from old Mk2 are wrong? But response is okay when I use node auth_and_test.

bugsounet, 2020-06-16:
of course, you can use it
credentials.json (AMk2) -> to MMM-GoogleAssistant directory
profiles/default.json (AMk2) -> to MMM-GoogleAssistant/token.json

bugsounet, 2020-06-16:
> does not react on anything i tried "ok google", "hey google", "smart mirror", mirror mirror on the wall"

by default the detector of snowboy is `jarvis`. You can change it in `snowboy {}` field
see [there](https://github.com/bugsounet/MMM-GoogleAssistant/wiki/Configuration-Structure#field-snowboy-) for more information

kffiatek, 2020-06-17:
Yes! "Jarvis" is the name! Great. Thanks! I'm sorry to bother you again but Assistant understands me, show responses on screen, but I can only hear "roger beep" when I call it. No voice answer. Did Google change something and voice responses doesn't work anymore?

bugsounet, 2020-06-17:
try this in terminal:
`dpkg -s pulseaudio | grep Status`
and give me the response

kffiatek, 2020-06-18:
Pulseaudio is not installed by default in Raspbian (or now Raspberry Pi OS). I followed your advice from 'installation' chapter, did not installed it. Sound works, even Assistant make this "roger beep" when I call it.

`pi@ak-mirror:~ $ dpkg -s pulseaudio | grep Status
dpkg-query: package 'pulseaudio' is not installed and no information is available
Use dpkg --info (= dpkg-deb --info) to examine archive files.`

lxne, 2020-07-03:
@kffiatek Did you find any solution for the audio output problem (chime sound yes, answer no)?
Because I once had it too. And there is another issue about it #74

kffiatek, 2020-07-03:
> 
> 
> @kffiatek Did you find any solution for the audio output problem (chime sound yes, answer no)?

Nope. Still waiting for the solution.

bugsounet, 2020-07-03:
Actually i have no solution.
why ?
GA use HTML5 audio (audio is sended by Chronium of MM)
there is no configuration for this.
I don't want to force playing output by `sh command` because in this case A2D don't works too (using HTML5 too)
If you want an assistant with output audio without HTML5:
try MMM-AssistantMk2 in my repository list

because, franckly, this issue will be never solved.
It's depend again of what do you have installed in your RPI in background in addition to MM (CPU/RAM usage)
I can determinate that sorry

lxne, 2020-07-03:
You can check the load of your cpu and memory by entering `top` in the terminal.

lxne, 2020-07-04:
Hi @bugsounet,

I do have one test installation (RPI3B+) with this issue.
If I ask GA something I get the chime and the animated symbol for voice output but no actual voice output.
'top' shows me that I have about 170MiB free memory.
When I first took a look I had only about 50MiB free so I decided to split less of the memory to the GPU. I did have set 128MB for the GPU. I changed it back to the default 64MB and after the reboot I had about 170MiB free when running MM and GA. Didn't change anything regarding the voice output problem.

Could there be any other explanation for that issue?

kffiatek, 2020-07-05:
It is unlikely that the problem is due to limited amount of memory. I do have this issue on RPi 4 with 8GB of RAM and almost nothing else running beside the MM. But I'll check it of course.

lxne, 2020-07-05:
> It is unlikely that the problem is due to limited amount of memory. I do have this issue on RPi 4 with 8GB of RAM and almost nothing else running beside the MM. But I'll check it of course.

What is your input and output hardware?

I realized that...
I am facing the no voice output problem when using my USB device as default input and output.
When I use the internal analog output (and the USB device only for input) it works without any problems.

bugsounet, 2020-07-10:
@kffiatek do you try lasted version with Native sound support ?

bugsounet, 2020-07-28:
Hi, I think your problem is now solved. reopen this issue if needed

closed 2020-07-28 by bugsounet