bugsounet/MMM-GoogleAssistant #43: helpp

8acoustic, 2020-06-14:
Welcome to MMM-GoogleAssistant

Checking OS...
OS Detected: linux-gnueabihf (raspbian 10 armv7l)

Checking all dependencies...
All Dependencies needed are installed !

Checking GCC Version...
Do you want to check compatible GCC version [Y/n] 
Your choice: y
You are using GCC 8, this is not compatible with this program.
You have to downgrade to GCC 7.
Do you want to make changes ? [Y/n] 
Your choice: 

should i put Y or N?

8acoustic, 2020-06-14:
i selected no
Your choice: n

Installing all npm libraries...
npm WARN deprecated request@2.88.2: request has been deprecated, see https://github.com/request/request/issues/3142

> grpc@1.23.4 install /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/grpc
> node-pre-gyp install --fallback-to-build --library=static_library

node-pre-gyp WARN Using request for node-pre-gyp https download 
[grpc] Success: "/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/grpc/src/node/extension_binary/node-v64-linux-arm-glibc/grpc_node.node" is installed via remote

> @bugsounet/snowboy@1.5.0 install /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy
> bash scripts/installer.sh

make: Entering directory '/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/build'
  CXX(target) Release/obj.target/snowboy/lib/snowboy.o
  SOLINK_MODULE(target) Release/obj.target/snowboy.node
  COPY Release/snowboy.node
  COPY /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/lib/node/binding/Release/node-v64-linux-arm/snowboy.node
  TOUCH Release/obj.target/action_after_build.stamp
make: Leaving directory '/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/build'

[@bugsounet/snowboy] Do you want to execute electron rebuild ? [Y/n]
Your choice: 
should i select no here?

bugsounet, 2020-06-15:
hi, all step are needed

PineMaster, 2020-06-15:
i think you need to downgrade

8acoustic, 2020-06-17:
/bio/bio.o
  CC(target) Release/obj.target/boringssl/deps/grpc/third_party/boringssl/crypto/bio/bio_mem.o
  CC(target) Release/obj.target/boringssl/deps/grpc/third_party/boringssl/crypto/bio/connect.o
../deps/grpc/third_party/boringssl/crypto/bio/connect.c: In function ‘conn_callback_ctrl’:
../deps/grpc/third_party/boringssl/crypto/bio/connect.c:491:29: error: cast between incompatible function types from ‘bio_info_cb’ {aka ‘long int (*)(struct bio_st *, int,  const char *, int,  long int,  long int)’} to ‘int (*)(const struct bio_st *, int,  int)’ [-Werror=cast-function-type]
       data->info_callback = (int (*)(const struct bio_st *, int, int))fp;
                             ^
cc1: all warnings being treated as errors
make: *** [boringssl.target.mk:434: Release/obj.target/boringssl/deps/grpc/third_party/boringssl/crypto/bio/connect.o] Error 1
make: Leaving directory '/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/grpc/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/node-gyp/lib/build.js:194:23)
gyp ERR! stack     at ChildProcess.emit (events.js:315:20)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:275:12)
gyp ERR! System Linux 4.19.97-v7+
gyp ERR! command "/usr/local/bin/node" "/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/.bin/node-gyp" "rebuild" "--target=6.1.7" "--arch=arm" "--dist-url=https://www.electronjs.org/headers" "--build-from-source" "--module_name=grpc_node" "--module_path=/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/grpc/src/node/extension_binary/electron-v6.1-linux-arm-glibc" "--host=https://node-precompiled-binaries.grpc.io/" "--remote_path={name}/v1.23.4" "--package_name=electron-v6.1-linux-arm-glibc.tar.gz"
gyp ERR! cwd /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/grpc
gyp ERR! node -v v12.18.0
gyp ERR! node-gyp -v v6.1.0
gyp ERR! not ok 

Failed with exit code: 1
    at SafeSubscriber._error (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/spawn-rx/lib/src/index.js:267:84)
    at SafeSubscriber.__tryOrUnsub (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/rxjs/internal/Subscriber.js:205:16)
    at SafeSubscriber.error (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/rxjs/internal/Subscriber.js:156:26)
    at Subscriber._error (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/rxjs/internal/Subscriber.js:92:26)
    at Subscriber.error (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/rxjs/internal/Subscriber.js:72:18)
    at MapSubscriber.Subscriber._error (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/rxjs/internal/Subscriber.js:92:26)
    at MapSubscriber.Subscriber.error (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/rxjs/internal/Subscriber.js:72:18)
    at SafeSubscriber._next (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/spawn-rx/lib/src/index.js:242:65)
    at SafeSubscriber.__tryOrUnsub (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/rxjs/internal/Subscriber.js:205:16)
    at SafeSubscriber.next (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/rxjs/internal/Subscriber.js:143:22)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! MMM-GoogleAssistant@2.0.1 postinstall: `installer/postinstall.sh`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the MMM-GoogleAssistant@2.0.1 postinstall script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2020-06-17T15_42_16_129Z-debug.log
pi@raspberrypi:~/MagicMirror/modules/MMM-GoogleAssistant $  

got this error while installation

bugsounet, 2020-06-17:
in your log ... i see
> gyp ERR! node -v v12.18.0

In [requirements](https://github.com/bugsounet/MMM-GoogleAssistant/wiki/Installation#requirements) it's v10.x

8acoustic, 2020-06-17:
okk thanks

8acoustic, 2020-06-17:
now its this : v10.16.0
iss this ver fine?

bugsounet, 2020-06-17:
lasted is v10 branch is:
```sh
pi@raspberrypi:~ $ node -v
v10.21.0
```
but it should works

8acoustic, 2020-06-17:
thanks hopefully this time it should work

8acoustic, 2020-06-17:
CC(target) Release/obj.target/boringssl/deps/grpc/third_party/boringssl/crypto/bio/connect.o
../deps/grpc/third_party/boringssl/crypto/bio/connect.c: In function ‘conn_callback_ctrl’:
../deps/grpc/third_party/boringssl/crypto/bio/connect.c:491:29: error: cast between incompatible function types from ‘bio_info_cb’ {aka ‘long int (*)(struct bio_st *, int,  const char *, int,  long int,  long int)’} to ‘int (*)(const struct bio_st *, int,  int)’ [-Werror=cast-function-type]
       data->info_callback = (int (*)(const struct bio_st *, int, int))fp;
                             ^
cc1: all warnings being treated as errors
make: *** [boringssl.target.mk:431: Release/obj.target/boringssl/deps/grpc/third_party/boringssl/crypto/bio/connect.o] Error 1
make: Leaving directory '/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/grpc/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/node-gyp/lib/build.js:194:23)
gyp ERR! stack     at ChildProcess.emit (events.js:198:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:248:12)
gyp ERR! System Linux 4.19.97-v7+
gyp ERR! command "/usr/local/bin/node" "/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/.bin/node-gyp" "rebuild" "--target=6.1.7" "--arch=arm" "--dist-url=https://www.electronjs.org/headers" "--build-from-source" "--module_name=grpc_node" "--module_path=/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/grpc/src/node/extension_binary/electron-v6.1-linux-arm-glibc" "--host=https://node-precompiled-binaries.grpc.io/" "--remote_path={name}/v1.23.4" "--package_name=electron-v6.1-linux-arm-glibc.tar.gz"
gyp ERR! cwd /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/grpc
gyp ERR! node -v v10.16.0
gyp ERR! node-gyp -v v6.1.0
gyp ERR! not ok 

Failed with exit code: 1
    at SafeSubscriber._error (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/spawn-rx/lib/src/index.js:267:84)
    at SafeSubscriber.__tryOrUnsub (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/rxjs/internal/Subscriber.js:205:16)
    at SafeSubscriber.error (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/rxjs/internal/Subscriber.js:156:26)
    at Subscriber._error (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/rxjs/internal/Subscriber.js:92:26)
    at Subscriber.error (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/rxjs/internal/Subscriber.js:72:18)
    at MapSubscriber.Subscriber._error (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/rxjs/internal/Subscriber.js:92:26)
    at MapSubscriber.Subscriber.error (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/rxjs/internal/Subscriber.js:72:18)
    at SafeSubscriber._next (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/spawn-rx/lib/src/index.js:242:65)
    at SafeSubscriber.__tryOrUnsub (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/rxjs/internal/Subscriber.js:205:16)
    at SafeSubscriber.next (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/rxjs/internal/Subscriber.js:143:22)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! MMM-GoogleAssistant@2.0.1 postinstall: `installer/postinstall.sh`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the MMM-GoogleAssistant@2.0.1 postinstall script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2020-06-17T17_19_13_024Z-debug.log
pi@raspberrypi:~/MagicMirror/modules/MMM-GoogleAssistant $ 

this is showing now

bugsounet, 2020-06-17:
retry with : 
`npm run rebuild` in `~/MagicMirror/modules/MMM-GoogleAssistant` directory

bugsounet, 2020-06-17:
and Yes to all

8acoustic, 2020-06-17:
im also getting this error: dpkg: error processing package libreoffice-help-common (--configure):

Do you want to check compatible GCC version [Y/n] 
Your choice: y
You are using GCC 8, this is not compatible with this program.
You have to downgrade to GCC 7.
Do you want to make changes ? [Y/n] 
Your choice: y
Installing GCC 7...
Reading package lists... Done
Building dependency tree       
Reading state information... Done
gcc-7 is already the newest version (7.3.0-19).
The following package was automatically installed and is no longer required:
  libmicrodns0
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 40 not upgraded.
3 not fully installed or removed.
Need to get 0 B/2,641 kB of archives.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] y
dpkg: error processing package libreoffice-help-common (--configure):
 package is in a very bad inconsistent state; you should
 reinstall it before attempting configuration
dpkg: dependency problems prevent configuration of libreoffice-help-en-us:
 libreoffice-help-en-us depends on libreoffice-help-common (= 1:6.1.5-3+rpi1+deb10u6+rpt1); however:
  Package libreoffice-help-common is not configured yet.

dpkg: error processing package libreoffice-help-en-us (--configure):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of libreoffice-help-en-gb:
 libreoffice-help-en-gb depends on libreoffice-help-common (= 1:6.1.5-3+rpi1+deb10u6+rpt1); however:
  Package libreoffice-help-common is not configured yet.

dpkg: error processing package libreoffice-help-en-gb (--configure):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 libreoffice-help-common
 libreoffice-help-en-us
 libreoffice-help-en-gb
E: Sub-process /usr/bin/dpkg returned an error code (1)

Installing all npm libraries...

> MMM-GoogleAssistant@2.0.1 postinstall /home/pi/MagicMirror/modules/MMM-GoogleAssistant
> installer/postinstall.sh

bugsounet, 2020-06-17:
ok, you have a mismake with `dpkg` and libreoffice
so my script can't downgrade to `gcc-7` and it's needed to compile grpc (google need gcc-7 version)

try to resolve you `dpkg` problem and retry
actually `apt-get update`, `apt-get upgrade` and `apt-get install`  commands are +/- disabled

8acoustic, 2020-06-17:
help pls fed up this error cant find any solution on google 

dpkg: error processing package libreoffice-help-en-gb (--configure):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 libreoffice-help-common
 libreoffice-help-en-us
 libreoffice-help-en-gb
E: Sub-process /usr/bin/dpkg returned an error code (1)

bugsounet, 2020-06-17:
maybe this:
```sh
sudo dpkg -P --force-all libreoffice-help-common libreoffice-help-en-us libreoffice-help-en-gb
```
**not sure**
if package removed like:
```
Removing libreoffice-help-common ...
Removing libreoffice-help-en-us ...
Removing libreoffice-help-en-gb...
```

try to update with:
 
```sh
sudo apt-get update
sudo apt-get upgrade
```

8acoustic, 2020-06-17:
pi@raspberrypi:~ $ sudo dpkg -P --force-all libreoffice-help-common libreoffice-help-en-us libreoffice-help-en-gb
dpkg: warning: overriding problem because --force enabled:
dpkg: warning: package is in a very bad inconsistent state; you should
 reinstall it before attempting a removal
(Reading database ... 143645 files and directories currently installed.)
Removing libreoffice-help-common (1:6.1.5-3+rpi1+deb10u6+rpt1) ...
dpkg: warning: ignoring request to remove libreoffice-help-en-us which isn't installed
dpkg: warning: ignoring request to remove libreoffice-help-en-gb which isn't installed
pi@raspberrypi:~ $ sudo apt-get update
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:2 http://archive.raspberrypi.org/debian buster InRelease
Hit:3 https://deb.nodesource.com/node_10.x buster InRelease
Reading package lists... Done
pi@raspberrypi:~ $ sudo apt-get upgrade
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
The following packages were automatically installed and are no longer required:
  ant ant-contrib ant-optional coinor-libcbc3 coinor-libcgl1 coinor-libclp1 coinor-libcoinmp1v5 coinor-libcoinutils3v5 coinor-libosi1v5 firebird3.0-common firebird3.0-common-doc firebird3.0-server-core firebird3.0-utils
  fonts-crosextra-caladea fonts-crosextra-carlito fonts-noto-core fonts-noto-ui-core fonts-opensymbol fonts-sil-gentium fonts-sil-gentium-basic gstreamer1.0-gtk3 libabw-0.1-1 libactivation-java libaopalliance-java libapache-poi-java
  libapache-pom-java libargs4j-java libasm-java libatinject-jsr330-api-java libbase-java libbcmail-java libbcpkix-java libbcprov-java libboost-chrono1.67.0 libboost-date-time1.67.0 libboost-iostreams1.67.0 libboost-locale1.67.0
  libbsh-java libcdi-api-java libcdr-0.1-1 libcglib-java libclucene-contribs1v5 libclucene-core1v5 libcmis-0.5-5v5 libcodemodel-java libcolamd2 libcommons-cli-java libcommons-codec-java libcommons-collections3-java
  libcommons-collections4-java libcommons-compress-java libcommons-io-java libcommons-lang3-java libcommons-logging-java libcommons-math3-java libcommons-parent-java libcurvesapi-java libdom4j-java libdtd-parser-java libe-book-0.1-1
  libehcache-java libel-api-java libeot0 libepubgen-0.1-1 libetonyek-0.1-1 libexttextcat-2.0-0 libexttextcat-data libfastinfoset-java libfbclient2 libflute-java libfonts-java libformula-java libfreehand-0.1-1
  libgeronimo-annotation-1.3-spec-java libgeronimo-interceptor-3.0-spec-java libgpgmepp6 libguava-java libguice-java libhawtjni-runtime-java libhsqldb1.8.0-java libhttpclient-java libhttpcore-java libib-util libicu4j-java
  libintellij-annotations-java libistack-commons-java libitext-java libjansi-java libjansi-native-java libjaxb-api-java libjaxb-java libjaxen-java libjcommon-java libjdom1-java libjetbrains-annotations-java libjsoup-java libjsp-api-java
  libjsr305-java liblangtag-common liblangtag1 liblayout-java libloader-java liblog4j1.2-java libmail-java libmaven-file-management-java libmaven-parent-java libmaven-resolver-java libmaven-shared-io-java libmaven-shared-utils-java
  libmaven3-core-java libmhash2 libmspub-0.1-1 libmwaw-0.3-3 libmythes-1.2-0 libneon27-gnutls libnumbertext-1.0-0 libnumbertext-data libodfgen-0.1-1 liborcus-0.14-0 libpagemaker-0.0-0 libpentaho-reporting-flow-engine-java libpixie-java
  libplexus-archiver-java libplexus-cipher-java libplexus-classworlds-java libplexus-component-annotations-java libplexus-interpolation-java libplexus-io-java libplexus-sec-dispatcher-java libplexus-utils2-java libpq5 libqxp-0.0-0
  libraptor2-0 librasqal3 librdf0 librelaxng-datatype-java libreoffice-common libreoffice-core libreoffice-style-colibre libreoffice-style-tango librepository-java librevenge-0.0-0 librngom-java libsac-java libsaxonhe-java
  libserializer-java libservlet-api-java libservlet3.1-java libsisu-guice-java libsisu-inject-java libsisu-ioc-java libsisu-plexus-java libslf4j-java libsnappy-java libsnappy-jni libstaroffice-0.0-0 libstax-ex-java libstreambuffer-java
  libsuitesparseconfig5 libtommath1 libtxw2-java libvisio-0.1-1 libwagon-http-java libwagon-provider-api-java libwebsocket-api-java libwpd-0.10-10 libwpg-0.3-3 libwps-0.4-4 libxerces2-java libxml-commons-external-java
  libxml-commons-resolver1.1-java libxml-java libxmlbeans-java libxmlsec1 libxmlsec1-nss libxom-java libxsom-java libxz-java libyajl2 libzmf-0.0-0 lp-solve node-normalize.css python3-uno uno-libs3 ure
Use 'sudo apt autoremove' to remove them.
The following packages have been kept back:
  python-rpi.gpio python3-rpi.gpio raspberrypi-ui-mods
0 upgraded, 0 newly installed, 0 to remove and 3 not upgraded.
pi@raspberrypi:~ $ 

got this 
is it fine now?

bugsounet, 2020-06-17:
maybe, just retry in GA directory
```sh
npm run rebuild
```

8acoustic, 2020-06-17:
GA is installed now mic hardware in not detecting its usb mic

Selection of the microphone device
**** List of CAPTURE Hardware Devices ****
Indicate the card # to use [0-9]: 9
you have selected: plughw:9
Checking audio input...
Make sure your microphone is on, press [Yes] and say something.
Press [No] if you don't want to check. [Y/n] 
Your choice: y
Current configuration not Working !

Selection of the microphone device
**** List of CAPTURE Hardware Devices ****
Indicate the card # to use [0-9]: 0
you have selected: plughw:0
Checking audio input...
Make sure your microphone is on, press [Yes] and say something.
Press [No] if you don't want to check. [Y/n] 
Your choice: n

MMM-GoogleAssistant is now installed !
Press [Enter] to continue

bugsounet, 2020-06-17:
if you have no mic detected, i can't help you

8acoustic, 2020-06-18:
ty for the help, now it works :)
can i link spotify to google assistant and play music using voice command?
im using MMM-NowPlayingOnSpotify

8acoustic, 2020-06-18:
can i activate voice assistant using groove gesture sensor???

bugsounet, 2020-06-18:
> im using MMM-NowPlayingOnSpotify
> can i activate voice assistant using groove gesture sensor???

not compatible with this module

8acoustic, 2020-06-24:
okayyy any other module for music which is compatible...
it would be really helpful to me

lxne, 2020-06-24:
> can i activate voice assistant using groove gesture sensor???

You can tell the assistant to listen or not listen to your configured hotword.
https://github.com/bugsounet/MMM-GoogleAssistant/wiki/Notifications

lxne, 2020-07-03:
> okayyy any other module for music which is compatible...
> it would be really helpful to me

Spotify integration is in the dev branch of MMM-Assistant2Display:
https://github.com/bugsounet/MMM-Assistant2Display/wiki/spotify

bugsounet, 2020-07-06:
hi,
main problem is now solved.
Reopen it if needed

closed 2020-07-06 by bugsounet