bugsounet/MMM-GoogleAssistant #44: User interface

lillaeriika, 2020-06-14:
Hey!

I'm not sure if I understand the position options in the config. Can you have anything else than fullscreen-above as position? Is there any way to have it like the old MK2, the response not being on top of all the other modules?

/Erik

lxne, 2020-06-14:
Good question. I tried a view things but nothing ever changed anything.

bugsounet, 2020-06-15:
Hi, now all position don't change ui, GA ui is in iframe.
It don't take any place in the mirror screen. It only showing when calling assistant.
I don't modify it sorry.
If you want an old ui interface, you can use AMk2 v3.5

Joshsamuel101, 2020-06-18:
Is there a way to modify the iFrame size so that the links are not full screen?

Had in mind something similar to this: https://github.com/l0b5ter/KeyDisplayed

bugsounet, 2020-06-18:
GA don't show any links, it's MMM-Assistant2Display job and it's displayed on Full screen size ([like this](https://raw.githubusercontent.com/bugsounet/MMM-Assistant2Display/dev/screenshot/screenshot2.png) with auto scrolling)
I don't know what can i do more

destiny1976, 2020-06-27:
is it possible to shrink the screensize of GA to 1x1 pixel at idle/listening mode? Is it possible only results are fullscreen?
I use a touchscreen, but when GA is idle, i can't use touchscreen anymore, because GA is always fullscreen on top.

bugsounet, 2020-06-28:
no, sorry... I don't code this ...
The module is an iframe for displying and have NO position in the screen to do that.

If you want this Feature, read and install AMk2 v3.5 repo (but not compatible with A2D)

closed 2020-07-06 by bugsounet