bugsounet/MMM-GoogleAssistant #45: Wake word not working for noisy environment

vivekpk007, 2020-06-16:
Hi, I was using MMM-AssistantMk2 with `with-AMk2v3-noisy_Jarvis.js`, it was working fine for me. After the update, the wake work is not working in a noisy environment
my config as follows 
``` {
  module: "MMM-GoogleAssistant",
  position: "fullscreen_above",
  config: {
    debug: false,
    assistantConfig: {
      lang: "en-US",
      projectId: "", // Required to use gaction.
      modelId: "", // (OPTIONAL for gaction)
      instanceId: "", // (OPTIONAL for gaction)
      latitude: 14.787940,
      longitude: 75.468556,
    },
    responseConfig: {
      useScreenOutput: true,
      screenOutputCSS: "screen_output.css",
      screenOutputTimer: 3000,
      activateDelay: 250,
      useAudioOutput: true,
      useChime: true,
      newChime: false
    },
    micConfig: { // put there configuration generated by auto-installer
      recorder: "arecord",
      device: "plughw:1",
    },
    customActionConfig: {
      autoMakeAction: false,
      autoUpdateAction: false, // in RPI, gaction CLI might have some trouble.(current version should be 2.2.4, but for linux-arm, Google haven't updated) so leave this as false in RPI. I don't know it is solved or not.
      actionLocale: "en-US", // At this moment, multi-languages are not supported, sorry. Someday I'll work.
    },
    snowboy: {
      audioGain: 2.0,
      Frontend: true,
      Model: "jarvis",
      Sensitivity: "null",
    },
    A2DServer: {
      useA2D: true,
      stopCommand: "stop"
    },
    recipes: [ "with-MMM-Spotify.js" ]
  }
```

bugsounet, 2020-06-16:
> After the update, the wake work is not working in a noisy environment

there is no noisy configuration in this module.
I don't know what do you want to say
please read [wiki](https://github.com/bugsounet/MMM-GoogleAssistant/wiki/Installation#auto-install-for-raspberrypi-or-debian-linux-machines-only) for installing
your wake word is defined in your `snowboy: {}` field

lavos-3rd, 2020-06-26:
What he meant was wake word cannot be detected when the area is noisy, mine for example is when music is playing, I keep saying "Alexa" in the microphone, even shouted at it, it won't respond

bugsounet, 2020-06-26:
that's why in snowboy field there is `Sensitivity` and you can ajust mic level too

bugsounet, 2020-07-06:
Hi, I think main issue is now solved,
Reopen it if needed

closed 2020-07-06 by bugsounet