bugsounet/MMM-GoogleAssistant #48: I cant install.

uok825, 2020-06-17:
I type npm install but when i got this error.
[https://prnt.sc/t1izqt](url)

bugsounet, 2020-06-17:
hi,
Just Read [requirements](https://github.com/bugsounet/MMM-GoogleAssistant/wiki/Installation#requirements)
You have raspbian 9

uok825, 2020-06-17:
> hi,
> Just Read [requirements](https://github.com/bugsounet/MMM-GoogleAssistant/wiki/Installation#requirements)
> You have raspbian 9
thats hurt :(

bugsounet, 2020-06-17:
Sorry, if you want to use it, you can try to `apt-get dist-upgrade`
It will upgrade your OS to lasted version

uok825, 2020-06-17:
> Sorry, if you want to use it, you can try to `apt-get dist-upgrade`
> It will upgrade your OS to lasted version

[https://prnt.sc/t1k59k](https://prnt.sc/t1k59k)

bugsounet, 2020-07-06:
hi,
main problem is now solved.
Reopen it if needed

closed 2020-07-06 by bugsounet