bugsounet/MMM-GoogleAssistant #53: View Additional info

PineMaster, 2020-06-18:
Hello, so i was wondering is there a way to view the links/articles that appear when I ask the assistant something.
also, how can I like extend the time it shows results and say it to stop showing results with voice commands?

bugsounet, 2020-06-18:
Hi, the better way is using [MMM-Assistant2Display](https://github.com/bugsounet/MMM-Assistant2Display)

Joshsamuel101, 2020-06-18:
> also, how can I like extend the time it shows results and say it to stop showing results with voice commands?

To extend time it shows results need to edit the MMM-GA config (- screenOutputTimer)
https://github.com/bugsounet/MMM-GoogleAssistant/wiki/Configuration-Structure
Default is 5 seconds

To stop showing results with voice commands use the A2D @bugsounet said above, when you've done that you need to reconfigure the MMM-GA config to reflect changed (-stopcommand, make sure -useA2D is true)
Deafult command is 'Stop' but I prefer 'Thank you'
Default is

bugsounet, 2020-07-06:
Hi,
main problem is now solved.
Reopen it if needed

closed 2020-07-06 by bugsounet