bugsounet/MMM-GoogleAssistant #54: Problem with configuration

TheLoopFlesh, 2020-06-18:
Hey bugsounet i had a issue. I had Raspberry Pi 4 model B with Respeaker 4 mic array from seeed. I install coretly, doing all steps  from (https://github.com/bugsounet/MMM-GoogleAssistant/wiki), and this didnt working. I had black screen on MagicMirror. Module didnt activate when im saing to him. Did u help me with VNC Viewer and skype?

lxne, 2020-06-18:
Whilest installing did you do `npm audit fix`?
Take a look at #41

Joshsamuel101, 2020-06-18:
> Hey bugsounet i had a issue. I had Raspberry Pi 4 model B with Respeaker 4 mic array from seeed. I install coretly, doing all steps from (https://github.com/bugsounet/MMM-GoogleAssistant/wiki), and this didnt working. I had black screen on MagicMirror. Module didnt activate when im saing to him. Did u help me with VNC Viewer and skype? I want to install GoogleAssistant with Assistant2D.

If this is the same issue I had (#41 ) then following the steps should fix it, I didn't have any issued with the A2D so that should be fine

TheLoopFlesh, 2020-06-19:
I do steps like in problem #41, i do npm run update, rebuild electron, and use npm audit fix and this is what i get. 

- [  run `npm audit fix` to fix them, or `npm audit` for details
pi@raspberrypi:~/MagicMirror/modules/MMM-GoogleAssistant $ npm audit fix

> grpc@1.24.3 install /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/grpc
> node-pre-gyp install --fallback-to-build --library=static_library

node-pre-gyp WARN Using request for node-pre-gyp https download 
[grpc] Success: "/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/grpc/src/node/extension_binary/node-v64-linux-arm-glibc/grpc_node.node" is installed via remote
+ grpc@1.24.3
removed 65 packages and updated 2 packages in 20.162s

2 packages are looking for funding
  run `npm fund` for details

fixed 4 of 6 vulnerabilities in 306 scanned packages
  2 vulnerabilities required manual review and could not be updated
pi@raspberrypi:~/MagicMirror/modules/MMM-GoogleAssistant $ npm fund
MMM-GoogleAssistant@2.0.2
├─┬ https://github.com/sponsors/isaacs
│ └── glob@7.1.6
└─┬ https://github.com/sponsors/sindresorhus
  └── p-limit@2.3.0
 ]

TheLoopFlesh, 2020-06-19:
And this is config.js from MagicMirror/config

/* Magic Mirror Config Sample
 *
 * By Michael Teeuw http://michaelteeuw.nl
 * MIT Licensed.
 *
 * For more information on how you can configure this file
 * See https://github.com/MichMich/MagicMirror#configuration
 *
 */

var config = {
	address: "localhost", // Address to listen on, can be:
	                      // - "localhost", "127.0.0.1", "::1" to listen on loopback interface
	                      // - another specific IPv4/6 to listen on a specific interface
	                      // - "0.0.0.0", "::" to listen on any interface
	                      // Default, when address config is left out or empty, is "localhost"
	port: 8080,
	ipWhitelist: ["127.0.0.1", "::ffff:127.0.0.1", "::1"], // Set [] to allow all IP addresses
	                                                       // or add a specific IPv4 of 192.168.1.5 :
	                                                       // ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.1.5"],
	                                                       // or IPv4 range of 192.168.3.0 --> 192.168.3.15 use CIDR format :
	                                                       // ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.3.0/28"],

	useHttps: false, 		// Support HTTPS or not, default "false" will use HTTP
	httpsPrivateKey: "", 	// HTTPS private key path, only require when useHttps is true
	httpsCertificate: "", 	// HTTPS Certificate path, only require when useHttps is true

	language: "en",
	timeFormat: 24,
	units: "metric",
	// serverOnly:  true/false/"local" ,
			     // local for armv6l processors, default
			     //   starts serveronly and then starts chrome browser
			     // false, default for all  NON-armv6l devices
			     // true, force serveronly mode, because you want to.. no UI on this device

	modules: [
		{
			module: "alert",
		},
{
  module: "MMM-GoogleAssistant",
  position: "fullscreen_above",
  config: {
    debug: false,
    assistantConfig: {
      lang: "en-US",
      projectId: "smartmirror-6d780", // Required to use gaction.
      modelId: "", // (OPTIONAL for gaction)
      instanceId: "", // (OPTIONAL for gaction)
      latitude: 51.508530,
      longitude: -0.076132,
    },
    responseConfig: {
      useScreenOutput: true,
      screenOutputCSS: "screen_output.css",
      screenOutputTimer: 5000,
      activateDelay: 250,
      useAudioOutput: true,
      useChime: true,
      newChime: false
    },
    micConfig: { // put there configuration generated by auto-installer
      recorder: "arecord",
      device: "plughw:2",
    },
    customActionConfig: {
      autoMakeAction: false,
      autoUpdateAction: false, // in RPI, gaction CLI might have some trouble.(current version should be 2.2.4, but for linux-arm, Google haven't updated) so leave this as false in RPI. I don't know it is solved or not.
      actionLocale: "en-US", // At this moment, multi-languages are not supported, sorry. Someday I'll work.
    },
    snowboy: {
      audioGain: 2.0,
      Frontend: true,
      Model: "jarvis",
      Sensitivity: 0.5
    },
    A2DServer: {
      useA2D: false,
      stopCommand: "stop"
    },
    recipes: [ "with-MMM-TelegramBot.js" ]
  }
},
		{
			module: "updatenotification",
			position: "top_bar"
		},
		{
			module: "clock",
			position: "top_left"
		},
		{
			module: "calendar",
			header: "US Holidays",
			position: "top_left",
			config: {
				calendars: [
					{
						symbol: "calendar-check",
						url: "webcal://www.calendarlabs.com/ical-calendar/ics/76/US_Holidays.ics"					}
				]
			}
		},
		{
			module: "compliments",
			position: "lower_third"
		},
		{
			module: "currentweather",
			position: "top_right",
			config: {
				location: "New York",
				locationID: "", //ID from http://bulk.openweathermap.org/sample/city.list.json.gz; unzip the gz file and find your city
				appid: "YOUR_OPENWEATHER_API_KEY"
			}
		},
		{
			module: "weatherforecast",
			position: "top_right",
			header: "Weather Forecast",
			config: {
				location: "New York",
				locationID: "5128581", //ID from http://bulk.openweathermap.org/sample/city.list.json.gz; unzip the gz file and find your city
				appid: "YOUR_OPENWEATHER_API_KEY"
			}
		},
		{
			module: "newsfeed",
			position: "bottom_bar",
			config: {
				feeds: [
					{
						title: "New York Times",
						url: "http://www.nytimes.com/services/xml/rss/nyt/HomePage.xml"
					}
				],
				showSourceTitle: true,
				showPublishDate: true,
				broadcastNewsFeeds: true,
				broadcastNewsUpdates: true
			}
		},
	]

};

/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== "undefined") {module.exports = config;}

lxne, 2020-06-19:
`npm audit fix` might cause the black screen problem.
That's why in #41 describes NOT to do `npm audit fix`.

bugsounet, 2020-06-19:
so try `npm run rebuild` in the GA directory

TheLoopFlesh, 2020-06-23:
Ok its working all good, but now i want to know how to add our Hotword from maybe snowboy.kitt.ai/dashoboard or others. I create the hotword and want to add him to config. I want to configurate hotword the same like google assistant on phone, he react only for my voice. U know how to do that?

bugsounet, 2020-06-25:
sorry, it's not yet implented and it's not planned actually

bugsounet, 2020-07-06:
Hi,
main problem is now solved.
Reopen it if needed

closed 2020-07-06 by bugsounet