bugsounet/MMM-GoogleAssistant #55: i cant find auth_and_test.js

MWHM212, 2020-06-18:
hi sir
when i try to execute this command auth_and_test.js it says no such file or directory

bugsounet, 2020-06-18:
what is the exact command do you write ?

MWHM212, 2020-06-18:
the same in wiki
cd ~/MagicMirror/modules/MMM-GoogleAssistant
then
node auth_and_test.js

bugsounet, 2020-06-18:
ok,
Can you try this:
```sh
cd ~/MagicMirror/modules/MMM-GoogleAssistant
ls -la
```
and give me the result plz

MWHM212, 2020-06-18:
total 32 drwxr-xr-x 4 pi pi 4096 Jun 18 15:20 . drwxr-xr-x 4 pi pi 4096 Jun 18 15:19 .. -rw-r--r-- 1 pi pi 430 Jun 18 15:13 credentials.json drwxr-xr-x 8 pi pi 4096 Jun 18 13:59 .git drwxr-xr-x 2 pi pi 4096 Jun 18 13:59 installer -rw-r--r-- 1 pi pi 1066 Jun 18 13:59 LICENSE -rw-r--r-- 1 pi pi 601 Jun 18 13:59 package.json -rw-r--r-- 1 pi pi 667 Jun 18 13:59 README.md

bugsounet, 2020-06-18:
hum, hum ... how to say ...
maybe it's better installing module before google setup no ?
well ... that's what the wiki says...

MWHM212, 2020-06-18:
which module do you mean ??

bugsounet, 2020-06-18:
Installation step missing...

MWHM212, 2020-06-18:
cd ~/MagicMirror/modules
git clone https://github.com/bugsounet/MMM-GoogleAssistant

cd MMM-GoogleAssistant
npm install

is there other step are not included here ??

bugsounet, 2020-06-18:
no but it's needed ... because it's install MMM-GoogleAssistant source.
you wrote:
```sh
total 32
drwxr-xr-x 4 pi pi 4096 Jun 18 15:20 .
drwxr-xr-x 4 pi pi 4096 Jun 18 15:19 ..
-rw-r--r-- 1 pi pi 430 Jun 18 15:13 credentials.json
drwxr-xr-x 8 pi pi 4096 Jun 18 13:59 .git
drwxr-xr-x 2 pi pi 4096 Jun 18 13:59 installer
-rw-r--r-- 1 pi pi 1066 Jun 18 13:59 LICENSE
-rw-r--r-- 1 pi pi 601 Jun 18 13:59 package.json
-rw-r--r-- 1 pi pi 667 Jun 18 13:59 README.md
```
so it's means ... you don't do this step
Actually, It's waiting to install command

arhumshahzad, 2020-06-19:
hey, i was having a similar problem. i followed the installation, upon using the command

cd ~/MagicMirror/modules/MMM-GoogleAssistant
node auth_and_test.js

i get 

internal/modules/cjs/loader.js:583
    throw err;
    ^

Error: Cannot find module 'google-assistant'
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:581:15)
    at Function.Module._load (internal/modules/cjs/loader.js:507:25)
    at Module.require (internal/modules/cjs/loader.js:637:17)
    at require (internal/modules/cjs/helpers.js:22:18)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/auth_and_test.js:7:25)
    at Module._compile (internal/modules/cjs/loader.js:689:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:700:10)
    at Module.load (internal/modules/cjs/loader.js:599:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:538:12)
    at Function.Module._load (internal/modules/cjs/loader.js:530:3)
i tried the command 

cd ~/MagicMirror/modules/MMM-GoogleAssistant
ls -la

total 96
drwxr-xr-x 9 pi pi  4096 Jun 19 04:12 .
drwxr-xr-x 5 pi pi  4096 Jun 19 04:11 ..
-rw-r--r-- 1 pi pi  2028 Jun 19 04:12 auth_and_test.js
drwxr-xr-x 2 pi pi  4096 Jun 19 04:12 components
-rw-r--r-- 1 pi pi   425 Jun 19 03:16 credentials.json
-rw-r--r-- 1 pi pi   132 Jun 19 04:12 deviceInstance.sample.json
drwxr-xr-x 8 pi pi  4096 Jun 19 04:12 .git
-rw-r--r-- 1 pi pi   145 Jun 19 04:12 .gitignore
drwxr-xr-x 2 pi pi  4096 Jun 19 04:12 installer
-rw-r--r-- 1 pi pi  1066 Jun 19 04:11 LICENSE
-rw-r--r-- 1 pi pi  3653 Jun 19 04:12 MMM-GoogleAssistant.css
-rw-r--r-- 1 pi pi 15854 Jun 19 04:12 MMM-GoogleAssistant.js
-rw-r--r-- 1 pi pi  6031 Jun 19 04:12 node_helper.js
-rw-r--r-- 1 pi pi  1083 Jun 19 04:12 package.json
-rw-r--r-- 1 pi pi   568 Jun 19 04:12 README.md
drwxr-xr-x 2 pi pi  4096 Jun 19 04:12 recipes
drwxr-xr-x 3 pi pi  4096 Jun 19 04:12 resources
-rw-r--r-- 1 pi pi  1339 Jun 19 04:12 screen_output.css
drwxr-xr-x 2 pi pi  4096 Jun 19 04:12 tmp
drwxr-xr-x 2 pi pi  4096 Jun 19 04:12 translations

If you have a solution id greatly appreciate it thanks

bugsounet, 2020-06-19:
```sh
pi@rpi:~$ cd ~/MagicMirror/modules
pi@rpi:~/MagicMirror/modules$ git clone https://github.com/bugsounet/MMM-GoogleAssistant
Clonage dans 'MMM-GoogleAssistant'...
remote: Enumerating objects: 119, done.
remote: Counting objects: 100% (119/119), done.
remote: Compressing objects: 100% (104/104), done.
remote: Total 513 (delta 45), reused 33 (delta 15), pack-reused 394
Réception d'objets: 100% (513/513), 2.78 MiB | 1.51 MiB/s, fait.
Résolution des deltas: 100% (206/206), fait.
pi@rpi:~/MagicMirror/modules$ cd MMM-GoogleAssistant
pi@rpi:~/MagicMirror/modules/MMM-GoogleAssistant$ npm install

> MMM-GoogleAssistant@ preinstall /home/bugs/MagicMirror/modules/MMM-GoogleAssistant
> installer/preinstall.sh

Welcome to MMM-GoogleAssistant

Checking OS...
OS Detected: linux-gnueabihf (raspbian 10 armv7l)

Checking all dependencies...
All Dependencies needed are installed !

Checking GCC Version...
Do you want to check compatible GCC version [Y/n] 
Your choice: Y
GCC 7 is set by default

Installing @sdetweil sandbox fix...
fixing sandbox permissions

Installing Sources...
La branche 'prod' est paramétrée pour suivre la branche distante 'prod' depuis 'origin'.
Déjà à jour.

Installing all npm libraries...
npm WARN deprecated request@2.88.2: request has been deprecated, see https://github.com/request/request/issues/3142

> grpc@1.23.4 install /home/bugs/MagicMirror/modules/MMM-GoogleAssistant/node_modules/grpc
> node-pre-gyp install --fallback-to-build --library=static_library

node-pre-gyp WARN Using request for node-pre-gyp https download 
[grpc] Success: "/home/bugs/MagicMirror/modules/MMM-GoogleAssistant/node_modules/grpc/src/node/extension_binary/node-v64-linux-x64-glibc/grpc_node.node" is installed via remote

> @bugsounet/snowboy@1.5.0 install /home/bugs/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy
> bash scripts/installer.sh

make : on entre dans le répertoire « /home/bugs/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/build »
  CXX(target) Release/obj.target/snowboy/lib/snowboy.o
  SOLINK_MODULE(target) Release/obj.target/snowboy.node
  COPY Release/snowboy.node
  COPY /home/bugs/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/lib/node/binding/Release/node-v64-linux-x64/snowboy.node
  TOUCH Release/obj.target/action_after_build.stamp
make : on quitte le répertoire « /home/bugs/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/build »

[@bugsounet/snowboy] Do you want to execute electron rebuild ? [Y/n]
Your choice: y
✔ Rebuild Complete

> protobufjs@6.9.0 postinstall /home/bugs/MagicMirror/modules/MMM-GoogleAssistant/node_modules/protobufjs
> node scripts/postinstall

> @bugsounet/snowboy@1.5.0 postinstall /home/bugs/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy
> tsc -p tsconfig.json

> MMM-GoogleAssistant@2.0.2 postinstall /home/bugs/MagicMirror/modules/MMM-GoogleAssistant
> installer/postinstall.sh

[grpc library]
Execute electron-rebuild...
It could takes 10~30 minutes.
✔ Rebuild Complete
Electron Rebuild Complete!

Checking Speaker and Microphone...
Do you want check your audio output configuration [Y/n] 
Your choice: n

MMM-GoogleAssistant is now installed !
Press [Enter] to continue

npm notice created a lockfile as package-lock.json. You should commit this file.
added 304 packages from 210 contributors and audited 306 packages in 334.653s

2 packages are looking for funding
  run `npm fund` for details

found 6 low severity vulnerabilities
  run `npm audit fix` to fix them, or `npm audit` for details

pi@rpi:~/MagicMirror/modules/MMM-GoogleAssistant$ node auth_and_test.js 
Opening OAuth URL. Return here with your code.
Paste your code: 
```
I detect no problem. I just do the installation part
I don't know what can i do more actually !?

MWHM212, 2020-06-19:
ok ,i will try to copy and paste the file 

can you paste the file here or if you want on gmail ??

bugsounet, 2020-06-19:
> ok ,i will try to copy and paste the file

what file ? there is no file to copy or paste.
I just execute install part and everything works fine
I past you the entire process in last post
After install part finish, I can access without error to `node auth_and_test.js`

arhumshahzad, 2020-06-19:
after redoing the installation process, when i get to "npm install" 

Installing all npm libraries...
Segmentation fault.] - fetchMetadata: sill pacote range manifest for node-html-

i get this and the one you posted had a different response, how do i fix this?

Ali229, 2020-07-02:
This does not work for windows:
```
D:\Program Files\MagicMirror\modules\MMM-GoogleAssistant>npm install

> MMM-GoogleAssistant@ preinstall D:\Program Files\MagicMirror\modules\MMM-GoogleAssistant
> installer/preinstall.sh

'installer' is not recognized as an internal or external command,
operable program or batch file.
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! MMM-GoogleAssistant@ preinstall: `installer/preinstall.sh`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the MMM-GoogleAssistant@ preinstall script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm WARN Local package.json exists, but node_modules missing, did you mean to install?

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\Ali\AppData\Roaming\npm-cache\_logs\2020-07-02T23_05_16_898Z-debug.log
```

bugsounet, 2020-07-03:
@Ali229 : what's the requirements ?
@arhumshahzad: verify npm version (requirements)

Ali229, 2020-07-03:
> @Ali229 : what's the requirements ?
> @arhumshahzad: verify npm version (requirements)

The magic mirror itself works for windows, that's why I asked. No need to be over smart. BYE!

MWHM212, 2020-07-09:
Hi
when i re download it , it takes more time but now it is working 
thank you all 
one last thing how can i start it 🤔😅

MWHM212, 2020-07-09:
Hi
when i re download it , it takes more time but now it is working 
thank you all

closed 2020-07-09 by MWHM212