bugsounet/MMM-GoogleAssistant #57: auth_and_test 'client_id' undefined

speedy2fast, 2020-06-20:
I'm following the steps for setting up/authorizing google assistant, but whenever I run "node auth_and_test.js" I get the following error

pi@raspberrypi:~/MagicMirror/modules/MMM-GoogleAssistant $ node auth_and_test.js 
/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/google-assistant/components/auth.js:27
  const oauthClient = new OAuth2(key.client_id, key.client_secret, key.redirect_uris[0]);
                                     ^

TypeError: Cannot read property 'client_id' of undefined
    at new Auth (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/google-assistant/components/auth.js:27:38)
    at new GoogleAssistant (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/google-assistant/index.js:21:16)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/auth_and_test.js:60:19)
    at Module._compile (internal/modules/cjs/loader.js:778:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
    at Module.load (internal/modules/cjs/loader.js:653:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
    at Function.Module._load (internal/modules/cjs/loader.js:585:3)
    at Function.Module.runMain (internal/modules/cjs/loader.js:831:12)
    at startup (internal/bootstrap/node.js:283:19)

I downloaded the client_secret_XXX.json and renamed it to credentials.json in the proper folder. I've been trying to work this out myself, but I just can't figure out what's causing this error to be thrown.

lavos-3rd, 2020-06-22:
Same here too

bugsounet, 2020-07-17:
yes, sometime google return a new format file. it's depend where you download credentials
I will fix this with my own new Google-Assistant npm library

actually i can't update with original Google-Assistant library because there is some bug with it on RPI

It's under coding ... and testing !
and It will be ready very soon (maybe this week)

bugsounet, 2020-08-01:
Must be fixed. Reopen issue if needed

closed 2020-08-01 by bugsounet