bugsounet/MMM-GoogleAssistant #58: Output audio

phungorquan, 2020-06-21:
Hi bugsounet,

I setup G.A and everything is OK, but i didn't heard anything from G.A
E.g: I said "jarvis" then i listened G.A setting up sound but after that, i said "tell me a story" i didn't hear anything eventhough it show a story description at the screen bottom. 
I used my 3D sound USB, and i test it independently. My sound is Ok. I also changed both config as this :https://github.com/bugsounet/MMM-GoogleAssistant/wiki/Configuration-Sample but it's still not working
Could you help me? :D

Thanks!
Best regards,

chanapai, 2020-06-23:
Same problem

lavos-3rd, 2020-06-24:
Same problem too

bugsounet, 2020-06-24:
hi, define your output device in your RPI desktop (volume icon) with VNC app

HTML5 audio is used for ALL modules (GA and A2D) and sub-module (A2D modules)
Minimum required is RPI3B+ (modules need 600 Mo RAM with GA/A2D) and raspbian buster (Raspberry Pi OS (32-bit) with desktop)
no installation of MM with `sudo` (root) and must be installed with pi user
After if pulseaudio is installed, this module not compatible with it
If you use your RPI for OTHER things it could be the problem (RAM issue)
I advise you to use it only with MagicMirror (no others things in background)

lxne, 2020-07-03:
> E.g: I said "jarvis" then i listened G.A setting up sound but after that, i said "tell me a story" i didn't hear anything eventhough it show a story description at the screen bottom.

Did you find a solution for this audio problem?
There are two other issues #74 and #42 reporting the same problem.
No concrete problem analysis or solution so far.

bugsounet, 2020-07-03:
RPI memory overload.
too many processes are running for the RPI.
he can't execute it
Happen when a lot of background program is installed in addition to MM and modules

I use HTML5 audio, that is to say audio playback by chronium directly without any configuration (like any internet browser)
There is no configuration on this

bugsounet, 2020-07-06:
hi,
GoogleAssistant i now V2.1.0 with native audio support
If problem retry with `useNative:true` in `responseConfig: {}` in your configuration file

closed 2020-07-06 by bugsounet

lxne, 2020-07-06:
>  GoogleAssistant i now V2.1.0 with native audio support

@bugsounet, thank you!

phungorquan, 2020-07-15:
I updated to new version and the response audio was working. Thanks. 
For someone know my config:

{
  module: "MMM-Assistant2Display",
  position: "top_left",
  config: {
    debug:false,
    useYoutube: true,
    links: {
      useLinks: true,
      displayDelay: 60 * 1000,
      scrollStep: 25,
      scrollInterval: 1000,
      scrollStart: 5000,
      scrollActivate: false,
      verbose: false
    },
    photos: {
      usePhotos: true,
      displayDelay: 10 * 1000
    }
  }
},

{
  module: "MMM-GoogleAssistant",
  position: "center", // fullscreen_above
  config: {
    debug: true,
    assistantConfig: {
      lang: "en-US",
      projectId: "", // Required to use gaction.
      modelId: "", // (OPTIONAL for gaction)
      instanceId: "", // (OPTIONAL for gaction)
      latitude: 51.508530,
      longitude: -0.076132,
    },
    responseConfig: {
      useScreenOutput: true,
      screenOutputCSS: "screen_output.css",
      screenOutputTimer: 5000,
      activateDelay: 250,
      useAudioOutput: true,
      useChime: true,
      newChime: false
    },
    micConfig: { // put there configuration generated by auto-installer
      recorder: "arecord",
      device: "plughw:2",
    },
    snowboy: {
      audioGain: 2.0,
      Frontend: true,
      Model: "jarvis",
      Sensitivity: null
    },
    A2DServer: {
      useA2D: true,
      stopCommand: "stop"
    }
  }
},

bugsounet, 2020-07-15:
when it want to works, it's great use :)

phungorquan, 2020-07-18:
Hello, i think i found out why my Youtube was crashed, and how to listen voice from G.A response.
[=======] FIRST [=======]
I will give you my config which could solve the problem as this topic. I connected speaker into jack 3.5 onboard of raspberry and used microphone via usb 3Dsound. 

my lsusb and aplay-l like this:

`pi@raspberrypi:~ $ lsusb
Bus 001 Device 004: ID 1b3f:2008 Generalplus Technology Inc. 
Bus 001 Device 005: ID 0424:7800 Standard Microsystems Corp. 
Bus 001 Device 003: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 002: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
pi@raspberrypi:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: b1 [bcm2835 HDMI 1], device 0: bcm2835 HDMI 1 [bcm2835 HDMI 1]
  Subdevices: 4/4
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
card 1: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 4/4
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
card 2: Device [USB Audio Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0`

I add this file '.asoundrc' in '/home/pi' like [this](https://www.youtube.com/watch?v=_VwIup3qhSg):
`pcm.!default {
  type asym
  playback.pcm {
    type plug
    slave.pcm "output"
  }
  capture.pcm {
    type plug
    slave.pcm "input"
  }
}
pcm.output {
        type hw
        card 1
}
ctl.!default {
        type hw
        card 1
}
pcm.input {
  type hw
  card 1
}`

And like [this](https://www.raspberrypi-spy.co.uk/2019/06/using-a-usb-audio-device-with-the-raspberry-pi/) in 'sudo nano /usr/share/alsa/alsa.conf'
`defaults.ctl.card 1
defaults.pcm.card 1`

And you can follow my "config.js" as the comment above.

[=======] SECOND [=======]
To avoid crash Youtube, i didn't know why but when i change to `card 2` in file ".asoundrc", Youtube was crashed almost time eventhough my hw as the `aplay -l` is 2 :).