bugsounet/MMM-GoogleAssistant #59: Can't get past the installation phase (MMM-GoogleAssistant)

kkongchi-maker, 2020-06-21:
```
`cd ~/MagicMirror/modules`
`git clone https://github.com/bugsounet/MMM-GoogleAssistant`

`cd MMM-GoogleAssistant`
`npm install`
```

everything is fine until
`npm install` 
then, this is what I'm met with 

```
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! MMM-GoogleAssistant@2.0.2 postinstall: `installer/postinstall.sh`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the MMM-GoogleAssistant@2.0.2 postinstall script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
```

Nodejs version is V10, using Raspbian 10 yet it still won't work.. any help would be appreciated

radokristof, 2020-06-21:
Try with sudo

bugsounet, 2020-06-22:
@radokristof why using sudo command ?

kkongchi-maker, 2020-06-22:
@radokristof I haven’t gotten around to trying sudo yet, but will update as soon as I do, @bugsounet do you have any other ideas on how to go about this?

bugsounet, 2020-06-22:
Just because you use sudo command for installing MagicMirror...
I don't know why using root permission for installing this...
I think it's better to read MagicMirror official website for installing
The installation of GA requires a clean installation of MagicMirror according to the installation website and not other internet tutorial.
I'm not going to code a huge script to check and repair a bad installation

radokristof, 2020-06-22:
@bugsounet I had similar issues (with a MagicMirror which is installed with the automated script - so it might be a correct install) that a postinstall script only runs when it is executed with sudo. I can't give an explanation for this, because I did not look into why it might be (I thought that npm does something - like writing to directories where the current user can't do this), but the error was almost the same... I don't say it will fix it for sure, just it might worth a try.

kkongchi-maker, 2020-06-22:
I didn’t realize it had prerequisites! So are you saying that the module didn’t install because I didn’t do a clean install of magic mirror? I hope I didn’t misunderstand. 

If so, I’ll try to uninstall magicmirror and do a clean install and then try to install this module.

kkongchi-maker, 2020-06-28:
> Just because you use sudo command for installing MagicMirror...
> I don't know why using root permission for installing this...
> I think it's better to read MagicMirror official website for installing
> The installation of GA requires a clean installation of MagicMirror according to the installation website and not other internet tutorial.
> I'm not going to code a huge script to check and repair a bad installation
> 
> Read [Pre-requirement](https://github.com/bugsounet/MMM-GoogleAssistant/wiki/Installation#pre-requirements)

kkongchi-maker, 2020-06-28:
@bugsounet Hello. As far as I'm concerned, I have downloaded the manual version of MagicMirror according to their instructions 
from here [https://docs.magicmirror.builders/getting-started/installation.html#manual-installation](url).
If it is not the way it is installed, what else could it be?

closed 2020-07-02 by kkongchi-maker