bugsounet/MMM-GoogleAssistant #61: Own hotword

lillaeriika, 2020-06-22:
Hi! I'm not really sure I understand. Is there a way to use your own hotword and not the preconfigured like in the old mk2-version?

lavos-3rd, 2020-06-22:
There's a section in the wiki that guides you in using snowboy.

lillaeriika, 2020-06-22:
Where? I can't find it in the wiki for the new one. Just in the old mk2 version.

nqthao12061997, 2020-06-23:
yeah i have the same problem as you. i can't file module MMM-Hotword

phungorquan, 2020-06-23:
This module didn't need hotword anymore, just download and use model as the wiki guideline

bugsounet, 2020-06-23:
This option is not implemented and will not be implemented.

* Why ?
kitt.ai website will be closed in 6 months, so I will not code custom keywords

* AMk2 ?
MMM-Hotword project is closed (too much memory consumption, slowdown of modules)
If you prefer AMk2, I won't hold you back.

* So ?
I remind you however that I make MMM-GoogleAssistant **for ME in priority**
I think this module is not made for you.

closed 2020-06-23 by bugsounet

lillaeriika, 2020-06-23:
Okay, the module is great. I'm really thankful for your work. I just wanted to make sure that i didn't miss anything.