bugsounet/MMM-GoogleAssistant #62: MMM-GoogleAssistant installation freezes

EHammer98, 2020-06-24:
I'am trying to install [MMM-GoogleAssistant](https://github.com/bugsounet/MMM-GoogleAssistant) on my Raspberry Pi Zero W without any luck.
The latest Rasbian is installed and i always preform a update & upgrade.
I have installed MM using this command: `yes | bash -c "$(curl -sL  https://raw.githubusercontent.com/sdetweil/MagicMirror_scripts/master/raspberry.sh)" `

After that i follow [this ](https://github.com/bugsounet/MMM-GoogleAssistant/wiki/Installation)instruction.
I already got some problems with: `npm install`.
It occurs when i put in "Y" when it asks: `[@bugsnounet/snowboy] Do you want to execute electron rebuild? [Y/n]`
I did it "rest" for like a day and then rebooted it by just removing the power (i know.. not good...).
I put in "Y" since the instructions say to rebuild: `Don't forget to do ALL electron-rebuild steps.`

After that it seems like everything is installed ok.
But when i continue the instrunction i get stuck again at: `cd ~/MagicMirror/modules/MMM-GoogleAssistant
node auth_and_test.js` (getting Google Assistant to work).
It gives me this error msg: `Illegal instruction`
the documentaion says to perfrom: `npm rebuild`.
It does it's thing and comes back without any error's. But when i try the "node" command again, i get the same error.

"npm rebuild" also sometimes get's my Pi frozen. I also tried: `npm rebuild --build-from-source`
but that will freeze my Pi as wel....

Note: If i follow [these ](https://developers.google.com/assistant/sdk/guides/library/python/embed/setup)instructions, it will work without any problem... But i want it to get to work with MagicMirror.
Any solutions?

lxne, 2020-06-24:
I don't think a Zero can handle GoogleAssistant:
https://github.com/bugsounet/MMM-GoogleAssistant/wiki/Installation#requirements

bugsounet, 2020-06-24:
I think it's not work too
note: electron-rebuild step take a lot of time for a RPI 3b+ / RPI 4 -> about 30 min !
so a pi zero will take a very very very long time (a couple of hours !)
It's not freeze just make electron-rebuild job ...
but i think, in final, it will not work sorry

EHammer98, 2020-06-24:
> I don't think a Zero can handle GoogleAssistant:
> https://github.com/bugsounet/MMM-GoogleAssistant/wiki/Installation#requirements

@lxne Thanks for the fast reply! Yeah i know that doesn't "support" a Zero, but the same goes for MM (as far as i can remeber...) And that runs just fine. I also tried Google Assistant standalone, so not as a MMM. See: `Note: If i follow these instructions, it will work without any problem... But i want it to get to work with MagicMirror.
Any solutions?` So that's why i don't really get it....

EHammer98, 2020-06-24:
> I think it's not work too
> note: electron-rebuild step take a lot of time for a RPI 3b+ / RPI 4 -> about 30 min !
> so a pi zero will take a very very very long time (a couple of hours !)
> It's not freeze just make electron-rebuild job ...
> but i think, in final, it will not work sorry

@bugsounet thank for the reply! Yeah i know it can take a while. So you say that the Zero is so slow that it just can't handle the rebuild? Is there also no other way around? (since Google Assistant it self will run just fine on a Zero)

bugsounet, 2020-06-24:
No way sorry
Actually MagicMirror + MMM-GoogleAssistant (without MMM-Assistant2Display) take 490 Mo Ram (when activating)
You have 512 Mo Ram in pi Zero. it's very difficult for it

EHammer98, 2020-06-24:
> No way sorry
> Actually MagicMirror + MMM-GoogleAssistant (without MMM-Assistant2Display) take 490 Mo Ram (when activating)
> You have 512 Mo Ram in pi Zero. it's very difficult for it

@bugsounet Ok thanks for the info. I will try in on a RPI 4B. I will report back and close it if this was the solution (i am sure it will)

EHammer98, 2020-06-29:
@bugsounet I can confirm that it indeed does work on the Pi 4 model b (2 and 4GB ram). Thanks! I do have another issue where it closes MM, starts a blank page and then closes it and start MM again when i say the trigger/hotword. But i think i need to put that issue under #67 So i will close this one for now.

closed 2020-06-29 by EHammer98