bugsounet/MMM-GoogleAssistant #63: MMM-GoogleAssistant create own hotword

TheLoopFlesh, 2020-06-25:
Hi guys, I want to know how to add my Hotword from maybe snowboy.kitt.ai/dashoboard or others. I create the hotword and I want to config it. I want to configurate hotword the same like google assistant on phone, It react only for my voice. U know how to do that?

bugsounet, 2020-06-25:
sorry, it's not yet implented and it's not planned actually

TheLoopFlesh, 2020-06-25:
Okey, so how to fix that default hotword like "Alexa" "Jarvis" dont react allways.

bugsounet, 2020-06-25:
verify your Mic level (in RPI desktop -> icon volume -> output device mixer)
and/or ajust `Sensitivity` level feature of snowboy. read [this part](https://github.com/bugsounet/MMM-GoogleAssistant/wiki/Configuration-Structure#field-snowboy-)

my personnal setting: 
 * mic level : 40%
 * Sensitivity of snowboy : 0.6

Attention : 
 * it's depend of your microphone type
 * everyone have not the same microphone
 * so ... settings are not the same for everyone
 * If you boost too much your mic level : you can have some fake positive keyword
 * in reverse, if mic level is too low ... it wil not reconize any keyword
 * if you boost too much your sensitivity of snowboy: you can have some fake positive keyword too
 * in reverse, if sensitivity is too low... it will not reconize your prefered keyword
 * ---> search the best value

For your information:
my keyword is `jarvis` and no issue for me.
Snowboy reconize my keyword at 5 meters without problem :)
Not many fake positive when there is many people in the room

So : 
 * in first, let's check your mic level
 * adjust sensitivity after

bugsounet, 2020-07-06:
Hi
Reopen this issue if needed

closed 2020-07-06 by bugsounet