bugsounet/MMM-GoogleAssistant #64: Register your Mirror as Google Assistant related device

ahmadmutzakhisyah, 2020-06-27:
Register your Mirror as a device related to Google Assistant, I already did this step but I did not understand the last step, I did not find the mirror application on the google home app, so how do I find it?, please help me

bugsounet, 2020-07-03:
it should appear on its own if the registration is successful.
Of course, google home app and Google SDK account should be the same

lxne, 2020-07-03:
> Register your Mirror as a device related to Google Assistant, I already did this step but I did not understand the last step, I did not find the mirror application on the google home app, so how do I find it?, please help me

What exactly do you mean with 'last step'. Can you link to the description of that step?

ahmadmutzakhisyah, 2020-07-04:
This works, and I have opened it, but I am confused about the function of the smart mirror on the Google Home app, what can be done when the Smart Mirror is already on the Google Home app, thank you very much

ahmadmutzakhisyah, 2020-07-04:
and one more I want to ask, why when google asisstant opened a website link on my smart mirror can't be displayed, how well the solution

ahmadmutzakhisyah, 2020-07-04:
> it should appear on its own if the registration is successful.
> Of course, google home app and Google SDK account should be the same

thank you for your help, i very happy with your project

bugsounet, 2020-07-04:
> and one more I want to ask, why when google asisstant opened a website link on my smart mirror can't be displayed, how well the solution

try to install `MMM-Assistant2Display` plugin (read wiki for the link)

ahmadmutzakhisyah, 2020-07-05:
> > and one more I want to ask, why when google asisstant opened a website link on my smart mirror can't be displayed, how well the solution
> 
> try to install `MMM-Assistant2Display` plugin (read wiki for the link)

I installed MMM-Assistant2Display
and I already put the code in config.js, but this still does not work I use the same configuration as the sample configuration, thank you in advance sorry to bother

bugsounet, 2020-07-06:
Hi, your main issue is now solved.
Read MMM-Assistant2Display wiki for links configuration.
And open an isssue if needed

closed 2020-07-06 by bugsounet