bugsounet/MMM-GoogleAssistant #65: Error when run installation script

SnyRo, 2020-06-28:
Hi all, I'm really new to this project and Github, so please bear with me.
I'm currently running MagicMirror on Raspbian Linux Debian 10 on VMware virtual machine.

When I ran auto install script of MMM-GoogleAssistant everything seem normal until I encounter this error and the installation stop: 
`Installing all npm libraries...`
`Segmentation fault.] -fetchMetadata: still install loadAllDepsIntoIdealTree`

How can I fix this?
Thank you for reading and really appreciate for helping!

bugsounet, 2020-06-28:
I already had this problem with raspbian on VMware.
I did not manage to resolve it.
try more debian 10 directly under WMware it works, I develop with that

SnyRo, 2020-06-28:
Thank you for your reply. So I decided to install debian 10 directly under VMWare, after installed MagicMirror successfully without in sudo command, I installed MMM-GoogleAssistant and get this error: 

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! MMM-GoogleAssistant@ preinstall: `installer/preinstall.sh`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the MMM-GoogleAssistant@ preinstall script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm WARN Local package.json exists, but node_modules missing, did you mean to install?

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/nguyen/.npm/_logs/2020-06-28T17_54_24_138Z-debug.log
nguyen@debian:~/MagicMirror/modules/MMM-GoogleAssistant$ npm install
npm WARN npm npm does not support Node.js v10.21.0
npm WARN npm You should probably upgrade to a newer version of node as we
npm WARN npm can't make any promises that npm will work with this version.
npm WARN npm Supported releases of Node.js are the latest release of 4, 6, 7, 8, 9.
npm WARN npm You can find the latest version at https://nodejs.org/

There is node_modules in MagicMirror's file, but the error found it missing. Please help me with this. Thank you!

bugsounet, 2020-06-28:
npm version seems to to outdated
npm v6.10.x and more is need

```sh
bugs@debian:~$ npm -v
6.14.4
bugs@debian:~$ sudo npm -g update
/usr/bin/npm -> /usr/lib/node_modules/npm/bin/npm-cli.js
/usr/bin/npx -> /usr/lib/node_modules/npm/bin/npx-cli.js
+ npm@6.14.5
updated 5 packages in 4.442s
bugs@debian:~$ npm -v
6.14.5
bugs@debian:~$ 
```

SnyRo, 2020-06-29:
Thank you for your reply.
I have installed the latest version of npm but noticed that node_modules and npm were in different directory: /usr/local/lib/node_modules/npm/bin/npm-cli.js and /usr/local/lib/node_modules/npm/bin/npx-cli.js :
`nguyen@debian:~$ which npm`
`/usr/local/bin/npm`

So I move it to /usr/local/lib: `mv /usr/local/lib/node_modules  /usr/lib/node_modules` then check the path again:
`nguyen@debian:~$ which npm`
`/usr/bin/npm`
And got npm-cli.js and npx-cli.js in the right directory like yours.

But I still have the same error when run install scripts. Do I have to uninstall npm globally then install it again with the path: `/usr/bin/npm`?

SnyRo, 2020-06-30:
Thank you for your reply! I figured out that the error is from debian not from the module. Sorrry for not posting the full errror earlier, my bad :(. I close the topic now.

closed 2020-06-30 by SnyRo