bugsounet/MMM-GoogleAssistant #66: Warn Main 828

osta43, 2020-06-28:
[2020-06-28 15:22:08.585] [LOG]    [SNOWBOY] Retry restarting...
[2020-06-28 15:22:10.657] [LOG]    [LPCM16] WARN: arecord: main:828: 
[2020-06-28 15:22:10.658] [LOG]    [LPCM16] WARN: Fehler beim Öffnen des Gerätes: Datei oder Verzeichnis nicht gefunden

this errorlog comes everytime i activate google assistant with the hotword

bugsounet, 2020-06-28:
> this errorlog comes everytime i activate google assistant with the hotword

with Hotword ?

SnyRo, 2020-06-30:
Hi, I encountered the same error like yours. Run `arecord -l` in terminal and see what is your main capture audio device. If it listed as device 0, go to config.js and and make change in micConfig field into `device: "plughw:0",`, make the same edit if it device 1 then `device: "plughw:1",`,etc...
Hope this can help, good luck!

bugsounet, 2020-07-06:
hi,
I think that your issue is now solved.
Reopen it if needed

closed 2020-07-06 by bugsounet