bugsounet/MMM-GoogleAssistant #68: hi.. my assitant is not working, I attach my file status in detail. pls help me

NightRiders22, 2020-06-29:
I installed it by referring to the wiki. There was no error message.

i read Google Assistant Setup, I proceeded in order and passed the part to test at the prompt

But it doesn't work to see if I missed Attach my file

do npm start

and magic mirror is working but google assitant is not working

plz help me...

I will wait for your reply every hour Thank you

[image]
[configjs.txt](https://github.com/bugsounet/MMM-GoogleAssistant/files/4845457/configjs.txt)
[MMM-GoogleAssistantjs.txt](https://github.com/bugsounet/MMM-GoogleAssistant/files/4845458/MMM-GoogleAssistantjs.txt)

NightRiders22, 2020-06-29:
my mic and speaker is good working

test in onlinemictest and youtube

bugsounet, 2020-06-29:
hi, in your config, you use
```		{
			  module: "MMM-AssistantMk2",
			  position: "top_right",
			  config: {
				useWelcomeMessage: "brief today",
			    coordinates: {
			      latitude: 51.5033640,
			      longitude: -0.1276250,
			    },
			    record: { // put there configuration generated by auto-installer
			      recorder : "arecord",
			      device: "plughw:1"
			    },
			  }
		},
```
and you don't use MMM-GoogleAssistant
I don't know what do you want to do with the modified file of MMM-GoogleAssistant
and what do you want to search to do but it's not possible with this method

May be the simple solution for you is using MMM-AssistantMk2

NightRiders22, 2020-06-29:
Thank you for your answer. I've solved it. While looking for a way, I think it was corrupted while referring to other settings.

I tried to give up after fixing this for 4 days

If you change Model: "jarvis", randomly, will it work?

bugsounet, 2020-06-29:
Sorry, i don't support helping of MMM-AssistantMk2 module
[Please read the wiki part of this module](https://github.com/bugsounet/MMM-AssistantMk2/wiki)

closed 2020-06-29 by bugsounet