bugsounet/MMM-GoogleAssistant #70: White screen appears in the magic mirror screen, including errors in the terminal

Mo7aaa, 2020-07-02:
[ERROR]  WARNING! Could not load config file. Starting with default configuration. Error found: Error: Failed to load gRPC binary module because it was not installed for the current system
Expected directory: electron-v6.1-linux-arm-glibc
Found: [node-v64-linux-arm-glibc]
This problem can often be fixed by running "npm rebuild" on the current system
Original error: Cannot find module '/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/grpc/src/node/extension_binary/electron-v6.1-linux-arm-glibc/grpc_node.node'
Require stack:
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/grpc/src/grpc_extension.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/grpc/src/client_interceptors.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/grpc/src/client.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/grpc/index.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/google-assistant/components/assistant.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/google-assistant/index.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/components/assistant.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_helper.js
- /home/pi/MagicMirror/js/app.js
- /home/pi/MagicMirror/js/electron.js
- /home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js
- 
[2020-07-02 03:10:53.213] [LOG]    Loading module helpers ...
[2020-07-02 03:10:53.214] [LOG]    No helper found for module: alert.
[2020-07-02 03:10:53.214] [LOG]    No helper found for module: clock.
[2020-07-02 03:10:53.215] [LOG]    Initializing new module helper ...
[2020-07-02 03:10:53.215] [LOG]    Module helper loaded: calendar
[2020-07-02 03:10:53.216] [LOG]    No helper found for module: compliments.
[2020-07-02 03:10:53.216] [LOG]    No helper found for module: currentweather.
[2020-07-02 03:10:53.217] [LOG]    No helper found for module: weatherforecast.
[2020-07-02 03:10:53.217] [LOG]    Initializing new module helper ...
[2020-07-02 03:10:53.218] [LOG]    Module helper loaded: newsfeed
[2020-07-02 03:10:53.218] [LOG]    Initializing new module helper ...
[2020-07-02 03:10:53.218] [LOG]    Module helper loaded: MMM-COVID19
[2020-07-02 03:10:53.230] [ERROR]  App threw an error during load
[2020-07-02 03:10:53.231] [ERROR]  Error: Failed to load gRPC binary module because it was not installed for the current system
Expected directory: electron-v6.1-linux-arm-glibc
Found: [node-v64-linux-arm-glibc]
This problem can often be fixed by running "npm rebuild" on the current system
Original error: Cannot find module '/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/grpc/src/node/extension_binary/electron-v6.1-linux-arm-glibc/grpc_node.node'
Require stack:
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/grpc/src/grpc_extension.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/grpc/src/client_interceptors.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/grpc/src/client.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/grpc/index.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/google-assistant/components/assistant.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/google-assistant/index.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/components/assistant.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_helper.js
- /home/pi/MagicMirror/js/app.js
- /home/pi/MagicMirror/js/electron.js
- /home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js
- 
    at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/grpc/src/grpc_extension.js:53:17)
    at Module._compile (internal/modules/cjs/loader.js:786:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:798:10)
    at Module.load (internal/modules/cjs/loader.js:645:32)
    at Function.Module._load (internal/modules/cjs/loader.js:560:12)
    at Module.require (internal/modules/cjs/loader.js:685:19)
    at require (internal/modules/cjs/helpers.js:16:16)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/grpc/src/client_interceptors.js:144:12)
    at Module._compile (internal/modules/cjs/loader.js:786:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:798:10)

bugsounet, 2020-07-02:
> code: 'MODULE_NOT_FOUND'

so retry with `npm run rebuild`, because the library is not installed

Mo7aaa, 2020-07-02:
@bugsounet npm 
notice created a lockfile as package-lock.json. You should commit this file.
added 304 packages from 210 contributors and audited 306 packages in 1002.795s

2 packages are looking for funding
  run `npm fund` for details

found 6 low severity vulnerabilities
  run `npm audit fix` to fix them, or `npm audit` for details

Mo7aaa, 2020-07-02:
the output when i run `npm fund` 
MMM-GoogleAssistant@2.0.3
├─┬ https://github.com/sponsors/isaacs
│ └── glob@7.1.6
└─┬ https://github.com/sponsors/sindresorhus
  └── p-limit@2.3.0

bugsounet, 2020-07-02:
and ? what does it mean ?

Mo7aaa, 2020-07-02:
i don't know what does fund mean or do, i tried to solve the 6 vulnerabilities using npm audit, still the same problem after npm run rebuild @bugsounet

bugsounet, 2020-07-02:
and so ... why using `npm audit fix` ? and again ... what does it mean ?

Mo7aaa, 2020-07-02:
because that was the output, after typing `npm run rebuild`, `npm audit fix` it should fix the vulnerabilities automatically, while `npm audit` it shows the list of libraries that has vulnerabilities also type some commands in order to solve the vulnerabilities, please help i need to run the mirror using google assistant, thanks in advance! @bugsounet

bugsounet, 2020-07-02:
ok and now ... shoud `npm audit fix` is on the command line to install or updating this module ?
For this response helping: see wiki -> installation OR wiki -> updating

Mo7aaa, 2020-07-02:
no it didn't fix the issue, okay i'm going to see the updating i have read the installation, thanks @bugsounet

closed 2020-07-05 by Mo7aaa