bugsounet/MMM-GoogleAssistant #71: when I called my google assistant, it came out itself

phihaidev, 2020-07-02:
This is when I exit with ctrl + Q

pi@raspberrypi:~ $ cd MagicMirror
pi@raspberrypi:~/MagicMirror $ npm start

> magicmirror@2.12.0 start /home/pi/MagicMirror
> DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js

[2020-07-02 15:31:32.990] [LOG]    Starting MagicMirror: v2.12.0
[2020-07-02 15:31:33.001] [LOG]    Loading config ...
[2020-07-02 15:31:33.013] [LOG]    Loading module helpers ...
[2020-07-02 15:31:33.016] [LOG]    No helper found for module: helloworld.
[2020-07-02 15:31:33.017] [LOG]    No helper found for module: alert.
[2020-07-02 15:31:33.106] [LOG]    Initializing new module helper ...
[2020-07-02 15:31:33.107] [LOG]    Module helper loaded: updatenotification
[2020-07-02 15:31:34.378] [LOG]    Initializing new module helper ...
[2020-07-02 15:31:34.380] [LOG]    Module helper loaded: calendar
[2020-07-02 15:31:34.382] [LOG]    No helper found for module: compliments.
[2020-07-02 15:31:34.384] [LOG]    No helper found for module: currentweather.
[2020-07-02 15:31:34.386] [LOG]    No helper found for module: weatherforecast.
[2020-07-02 15:31:34.470] [LOG]    Initializing new module helper ...
[2020-07-02 15:31:34.472] [LOG]    Module helper loaded: newsfeed
[2020-07-02 15:31:34.474] [LOG]    No helper found for module: MMM-Snow.
[2020-07-02 15:31:34.476] [LOG]    No helper found for module: MMM-OClock.
[2020-07-02 15:31:36.969] [LOG]    Initializing new module helper ...
[2020-07-02 15:31:36.970] [LOG]    Module helper loaded: MMM-GoogleAssistant
[2020-07-02 15:31:36.970] [LOG]    All module helpers loaded.
[2020-07-02 15:31:37.160] [LOG]    Starting server on port 8080 ... 
[2020-07-02 15:31:37.196] [LOG]    Server started ...
[2020-07-02 15:31:37.199] [LOG]    Connecting socket for: updatenotification
[2020-07-02 15:31:37.201] [LOG]    Connecting socket for: calendar
[2020-07-02 15:31:37.203] [LOG]    Starting node helper for: calendar
[2020-07-02 15:31:37.205] [LOG]    Connecting socket for: newsfeed
[2020-07-02 15:31:37.206] [LOG]    Starting node helper for: newsfeed
[2020-07-02 15:31:37.208] [LOG]    Connecting socket for: MMM-GoogleAssistant
[2020-07-02 15:31:37.210] [LOG]    Sockets connected & modules started ...
[2020-07-02 15:31:37.502] [LOG]    Launching application.
[2020-07-02 15:31:43.316] [INFO]   Newsfeed-Fetcher: Broadcasting 56 items.
[2020-07-02 15:31:44.589] [INFO]   Newsfeed-Fetcher: Broadcasting 50 items.
[2020-07-02 15:31:44.722] [INFO]   Newsfeed-Fetcher: Broadcasting 10 items.
[2020-07-02 15:31:44.911] [INFO]   Newsfeed-Fetcher: Broadcasting 30 items.
[2020-07-02 15:31:44.970] [INFO]   Checking git for module: MMM-Snow
[2020-07-02 15:31:45.062] [INFO]   Checking git for module: MMM-OClock
[2020-07-02 15:31:45.133] [INFO]   Checking git for module: MMM-GoogleAssistant
[2020-07-02 15:31:45.409] [INFO]   Calendar-Fetcher: Broadcasting 1 events.
[2020-07-02 15:31:46.382] [INFO]   Newsfeed-Fetcher: Broadcasting 56 items.
[2020-07-02 15:31:46.522] [LOG]    [ASSISTANT] MMM-GoogleAssistant Version: 2.0.3
[2020-07-02 15:31:46.541] [LOG]    [SNOWBOY] Model selected: jarvis
[2020-07-02 15:31:46.566] [LOG]    [SNOWBOY] snowboy v1.5.0 Initialized...
[2020-07-02 15:31:46.570] [LOG]    [ASSISTANT] Assistant2Display Server Started
[2020-07-02 15:31:46.571] [LOG]    [ASSISTANT] Google Assistant is initialized.
[2020-07-02 15:31:46.702] [LOG]    Use existing calendar fetcher for url: https://www.calendarlabs.com/ical-calendar/ics/77/Vietnam_Holidays.ics
[2020-07-02 15:31:46.705] [INFO]   Calendar-Fetcher: Broadcasting 1 events.
[2020-07-02 15:31:47.072] [LOG]    Use existing calendar fetcher for url: https://www.calendarlabs.com/ical-calendar/ics/77/Vietnam_Holidays.ics
[2020-07-02 15:31:47.073] [INFO]   Calendar-Fetcher: Broadcasting 1 events.
[2020-07-02 15:31:47.253] [INFO]   Newsfeed-Fetcher: Broadcasting 50 items.
[2020-07-02 15:31:47.326] [INFO]   Newsfeed-Fetcher: Broadcasting 10 items.
[2020-07-02 15:31:47.773] [INFO]   Newsfeed-Fetcher: Broadcasting 30 items.
[2020-07-02 15:31:47.989] [LOG]    Use existing calendar fetcher for url: https://www.calendarlabs.com/ical-calendar/ics/77/Vietnam_Holidays.ics
[2020-07-02 15:31:47.991] [INFO]   Calendar-Fetcher: Broadcasting 1 events.
[2020-07-02 15:31:48.261] [LOG]    Use existing calendar fetcher for url: https://www.calendarlabs.com/ical-calendar/ics/77/Vietnam_Holidays.ics
[2020-07-02 15:31:48.263] [INFO]   Calendar-Fetcher: Broadcasting 1 events.
[2020-07-02 15:31:48.325] [INFO]   Calendar-Fetcher: Broadcasting 1 events.
[2020-07-02 15:31:48.407] [INFO]   Calendar-Fetcher: Broadcasting 1 events.
[2020-07-02 15:31:48.651] [LOG]    Use existing calendar fetcher for url: https://www.calendarlabs.com/ical-calendar/ics/77/Vietnam_Holidays.ics
[2020-07-02 15:31:48.653] [INFO]   Calendar-Fetcher: Broadcasting 1 events.
[2020-07-02 15:31:48.944] [INFO]   Newsfeed-Fetcher: Broadcasting 56 items.
[2020-07-02 15:31:49.061] [LOG]    Use existing calendar fetcher for url: https://www.calendarlabs.com/ical-calendar/ics/77/Vietnam_Holidays.ics
[2020-07-02 15:31:49.063] [INFO]   Calendar-Fetcher: Broadcasting 1 events.
[2020-07-02 15:31:49.207] [INFO]   Calendar-Fetcher: Broadcasting 1 events.
[... 184 lines cut ...]
[2020-07-02 15:35:00.250] [LOG]    Use existing calendar fetcher for url: https://www.calendarlabs.com/ical-calendar/ics/77/Vietnam_Holidays.ics
[2020-07-02 15:35:00.252] [INFO]   Calendar-Fetcher: Broadcasting 1 events.
[2020-07-02 15:35:00.548] [LOG]    Use existing calendar fetcher for url: https://www.calendarlabs.com/ical-calendar/ics/77/Vietnam_Holidays.ics
[2020-07-02 15:35:00.550] [INFO]   Calendar-Fetcher: Broadcasting 1 events.
[2020-07-02 15:35:00.575] [INFO]   Newsfeed-Fetcher: Broadcasting 56 items.
[2020-07-02 15:35:00.770] [INFO]   Newsfeed-Fetcher: Broadcasting 30 items.
[2020-07-02 15:35:00.920] [INFO]   Calendar-Fetcher: Broadcasting 1 events.
[2020-07-02 15:35:00.925] [LOG]    Use existing calendar fetcher for url: https://www.calendarlabs.com/ical-calendar/ics/77/Vietnam_Holidays.ics
[2020-07-02 15:35:00.926] [INFO]   Calendar-Fetcher: Broadcasting 1 events.
[2020-07-02 15:35:01.310] [INFO]   Newsfeed-Fetcher: Broadcasting 50 items.
[2020-07-02 15:35:01.478] [INFO]   Calendar-Fetcher: Broadcasting 1 events.
[2020-07-02 15:35:01.598] [INFO]   Newsfeed-Fetcher: Broadcasting 10 items.
/home/pi/MagicMirror/node_modules/electron/dist/electron js/electron.js: symbol lookup error: /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/grpc/src/node/extension_binary/electron-v6.1-linux-arm-glibc/grpc_node.node: undefined symbol: __atomic_store_8
npm ERR! code ELIFECYCLE
npm ERR! syscall spawn
npm ERR! file sh
npm ERR! errno ENOENT
npm ERR! magicmirror@2.12.0 start: `DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js`
npm ERR! spawn ENOENT
npm ERR! 
npm ERR! Failed at the magicmirror@2.12.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2020-07-02T08_35_01_919Z-debug.log

bugsounet, 2020-07-02:
hi, just Read Wiki -> Updating for solve this

phihaidev, 2020-07-02:
I tried many times, even raspbian reinstall but when I called my assistant, magicmirror escaped on its own.

bugsounet, 2020-07-02:
> undefined symbol: __atomic_store_8

It's means: you have updated gprc google package to 1.24.x and ONLY 1.23.4 is needed
Now... if you have tried `many times` and reinstall `raspbian` (many times too) ...
It's just means don't use this module, because it's not compatible to your system

closed 2020-07-03 by phihaidev