bugsounet/MMM-GoogleAssistant #73: MMM-GoogleAssistant is not responding

kkongchi-maker, 2020-07-02:
GA is not responding to "jarvis", "snowboy", "computer", "alex".

1. I have configured config.js accordingly to the guide
2. adjusted the volume icon
3. edited the .asoundrc file

everything looks fine yet GA does not respond. if there are any errors causing this problem, I would be grateful.

here's the card and device numbers

```
~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: b1 [bcm2835 HDMI 1], device 0: bcm2835 HDMI 1 [bcm2835 HDMI 1]
  Subdevices: 4/4
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
card 1: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 4/4
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
~ $ arecord -l
**** List of CAPTURE Hardware Devices ****
card 2: USB [Realtek Audio USB], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

```

so I'm assuming 
headphones
card 1 device 0

mic
card 2 device 0.
please correct me if I'm wrong.

kkongchi-maker, 2020-07-02:
here's the config file that I pasted unto a txt file
[config js.txt](https://github.com/bugsounet/MMM-GoogleAssistant/files/4866641/config.js.txt)

audio is fine (input and output)
[image]

.asoundrc file that i edited 
[image]

when I do a mic and audio check with this : 
1. Record for microphone test
`arecord --format=S16_LE --duration=5 --rate=16000 --file-type=raw out.raw`

2. Play the recorded file with the application
`aplay --format=S16_LE --rate=16000 out.raw`

everything works out fine, its just the module it seems. any help would be appreciated

bugsounet, 2020-07-02:
so, if the result of arecord is that
change your device to:

```js
device: "plughw:2"
```

Note : auto-installer can't never done this result : `device: "plughw:2,0"`

kkongchi-maker, 2020-07-02:
[image]

It still does not work.

bugsounet, 2020-07-02:
do you use another record to do another thing ?
because you can only use one recorder at the time (with alsa)

kkongchi-maker, 2020-07-02:
no I am using only 1 mic.

could it be that I edited the MMM-GoogleAssistant.js file before and somehow corrupted it? i will attach the txt file here
[mmm-googleassistant js.txt](https://github.com/bugsounet/MMM-GoogleAssistant/files/4866816/mmm-googleassistant.js.txt)

lxne, 2020-07-03:
> could it be that I edited the MMM-GoogleAssistant.js file

What exactly did you edit and why?
Try using the original file.

bugsounet, 2020-07-03:
> Try using the original file.

for restoring the original file:
```sh
cd ~/MagicMirror/modules/MMM-GoogleAssistant
rm MMM-GoogleAssistant.js && git checkout MMM-GoogleAssistant.js
```

and retry launching MM with : 
```sh
cd ~/MagicMirror
npm start
```

and give logs

kkongchi-maker, 2020-07-05:
@lxne I edited the MMM-GoogleAssistant.js file on /home/pi/MagicMirror/modules/MMM-GoogleAssistant
I did it because I thought the configuration had to be done on individual files instead of the config.js file on /home/pi/MagicMirror/config
I changed the language, latitude, longitude, simple things like that

the output of arecord -l is 
```
~ $ arecord -l
**** List of CAPTURE Hardware Devices ****
card 2: USB [Realtek Audio USB], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

I will try to use the og file. 

thank you

kkongchi-maker, 2020-07-05:
@bugsounet thank you. I will give that a shot.

kkongchi-maker, 2020-07-05:
I'm sorry I'm new to this, how do I show you my logs?

Also,  I restored the original file and it worked, but only once. It responded to me when I said snowboy, what time is it but not when I called the 2nd time.

[image]
did I configure this correctly?

kkongchi-maker, 2020-07-05:
thank you so much! it is working now. the problem was that I messed with the MMM-GoogleAssistant.js file, that is why it was not working

how can I fix the misalignment of the text in the google assistant visual?
[image]

bugsounet, 2020-07-06:
It seems to be an CSS problem
 * maybe screen resolution too low ?
 * try to upgrade GA to lastest version (v2.1.0)
 * try another position fullscreen_above -> top_left (it's another CSS ui)

bugsounet, 2020-07-21:
in all case main issue is now solved. open another issue if problem

closed 2020-07-21 by bugsounet