bugsounet/MMM-GoogleAssistant #74: Chime sound and responding, but no voice

EHammer98, 2020-07-02:
I have a new issue,
I am now using a USB sound card and works great i can listen to music, no problem. (it worked before without any problems with audio trough HDMI and a dedicated USB mic)
the microfoon input also works great.
but when start MM with GoogleAssistant and say "Jarvis"
I hear the chime, i ask my question (about the weather for example)
And then i get feedback, but only in text/image form. No verbal voice.

Somebody experience with that?

Mo7aaa, 2020-07-03:
You should thank god 🤲, i have ana issue after adding the google assistant configuration in the config file of the MM nothing works just a black screen also other modules doesn't work 🙂 @EHammer98

EHammer98, 2020-07-03:
> You should thank god 🤲, i have ana issue after adding the google assistant configuration in the config file of the MM nothing works just a black screen also other modules doesn't work 🙂 @EHammer98

I had that too, did you use the sample configuration? Because that should work, if it doesn't try to install it again and rebuild it again

lxne, 2020-07-03:
> You should thank god 🤲, i have ana issue after adding the google assistant configuration in the config file of the MM nothing works just a black screen also other modules doesn't work 🙂 @EHammer98

Do not do any 'npm audit fix' when installing GA.
That solved the black screen issue for me.

lxne, 2020-07-03:
> I have a new issue,
> I am now using a USB sound card and works great i can listen to music, no problem. (it worked before without any problems with audio trough HDMI and a dedicated USB mic)
> the [...]

I had the exact same problem. Unfortunately only a fresh installation solved it for me.

EHammer98, 2020-07-03:
> I had the exact same problem. Unfortunately only a fresh installation solved it for me.

Thanks for the reply. Fresh installation of Raspbian?, MM or only GA?

lxne, 2020-07-03:
> Thanks for the reply. Fresh installation of Raspbian?, MM or only GA?

I did it all from scratch. But maybe fresh MM would be sufficient.

EHammer98, 2020-07-03:
> > Thanks for the reply. Fresh installation of Raspbian?, MM or only GA?
> 
> I did it all from scratch. But maybe fresh MM would be sufficient.

Ok thx! I will try to remove my MM folder and reinstall it

Mo7aaa, 2020-07-03:
Yeah , i have used the sample configuration is it wrong ?? @EHammer98

Mo7aaa, 2020-07-03:
I'm going to install it without `npm audit fix` thanks hope it works, but also it shows errors in the terminal, please could you check my issue, thanks again @lxne

lxne, 2020-07-03:
> I'm going to install it without `npm audit fix` thanks hope it works, but also it shows errors in the terminal, please could you check my issue, thanks again @lxne

What errors?

EHammer98, 2020-07-03:
> > > Thanks for the reply. Fresh installation of Raspbian?, MM or only GA?
> > 
> > 
> > I did it all from scratch. But maybe fresh MM would be sufficient.
> 
> Ok thx! I will try to remove my MM folder and reinstall it

Nop it didn't work. Tried MM reinstall and also complete clean install of Raspbian

EHammer98, 2020-07-03:
> Yeah , i have used the sample configuration is it wrong ?? @EHammer98

No that should work

lxne, 2020-07-03:
> Nop it didn't work. Tried MM reinstall and also complete clean install of Raspbian

Still only the chime and nothing else?

EHammer98, 2020-07-03:
> > Nop it didn't work. Tried MM reinstall and also complete clean install of Raspbian
> 
> Still only the chime and nothing else?

Yeah, only chime... I do get feedback. But just no voice.... Normal sound, like playing music via the webbrowser does also work. Just the GA voice that i don't hear :/

Mo7aaa, 2020-07-03:
any advice which configuration should i use sample or the full ? @EHammer98

EHammer98, 2020-07-03:
> any advice which configuration should i use sample or the full ? @EHammer98

Sample

lxne, 2020-07-05:
> Yeah, only chime... I do get feedback. But just no voice....

What is your default output device?
Did you try to use the internal analog output as default?
That worked for me. USB output did not work (only chime), internal output worked.

majik655, 2020-07-06:
> 
> 
> > Yeah, only chime... I do get feedback. But just no voice....
> 
> What is your default output device?
> Did you try to use the internal analog output as default?
> That worked for me. USB output did not work (only chime), internal output worked.

Thank you... now she speaks!  I was using a usb soundcard that had microphone and speaker but I just had no sound from her except chimes and dings...fixed as soon as I went to analog on board sound
Thank you

bugsounet, 2020-07-06:
hi,
GoogleAssistant i now V2.1.0 with native audio support
If problem retry with `useNative:true` in `responseConfig: {}` in your configuration file

lxne, 2020-07-06:
> If problem retry with `useNative:true` in `responseConfig: {}` in your configuration file

Thank you @bugsounet. Works great.

bugsounet, 2020-07-06:
sure but ... I didn't want to code this at the start ... (only in HTML5 ... like youtube API, my Cast API, and others)

bugsounet, 2020-07-06:
So ... now @lxne we should have another problem in A2D ... let's wait and see !

closed 2020-07-12 by bugsounet

EHammer98, 2020-07-27:
@bugsounet sorry for the very late response. But thx! It did work with your solution

masoudl, 2020-09-05:
I had to try changing the player program to get one that behaves correctly. Even though I have mpg321 and aplay and play of others, the one that worked properly was "play".