bugsounet/MMM-GoogleAssistant #75: how to set location of google assistant on magic mirror

phihaidev, 2020-07-04:
I live in Vietnam, but when I ask my google assistant about time and weather, my google assistant answers London's information, how can I install google assistant to answer it correctly? Time and weather information from my location in real time, thank very much

bugsounet, 2020-07-04:
Hi, you can determinate your localisation in your module configuration from your config.js file
Read wiki for more informations (configuration sample and structure)

phihaidev, 2020-07-05:
Thank you very much, I tried and succeeded when it responded to the time and weather right to where I am now, but I don't know how I did it, I tried it and succeeded it answered time. and the weather is right for my current location, but I don't know how to play a song on the magic mirror with google assistant, I installed MMM-GoogleAssistant and MMM-Assistant2Display

bugsounet, 2020-07-06:
Hi perfect so ;)

For song music, read Assistant2Display wiki.
actually, Only youtube and spotify is supported

Please, open an issue in A2D if needed

closed 2020-07-06 by bugsounet