bugsounet/MMM-GoogleAssistant #76: Small help with Google Assistant response

majik655, 2020-07-05:
I have finally got this Google Assistant module working ...
I am on brand new installed Buster image and newest magic mirror (no other modules except defaults).

Currently when I say Jarvis I hear the confirm beeps and chimes and GA comes up full screen and is responding with text... however am I supposed to hear the GA voice??

Is there a setting I am missing somewhere ?
Everything is running perfect I just don't hear a response from GA voice .

```
		{
  			module: "MMM-GoogleAssistant",
  			position: "fullscreen_above",
  			config: {
    				debug: false,
    				assistantConfig: {
      						lang: "en-US",
      						projectId: "MYPROJECTIDISHERE", // Required to use gaction.
      						modelId: "MYMODELIDISHERE", // (OPTIONAL for gaction)
      						instanceId: "", // (OPTIONAL for gaction)
      						latitude: 45.857422,
      						longitude: -122.828377,
    						},
    				responseConfig: {
      						useScreenOutput: true,
      						screenOutputCSS: "screen_output.css",
      						screenOutputTimer: 5000,
      						activateDelay: 250,
      						useAudioOutput: true,
      						useChime: true,
      						newChime: false
    						},
    				micConfig: { // put there configuration generated by auto-installer
      						recorder: "arecord",
      						device: "plughw:2",
    					   },
    				customActionConfig: {
      							autoMakeAction: false,
      							autoUpdateAction: false, // in RPI, gaction CLI might have some trouble.(current version should be 2.2.4, but for linux-arm, Google haven't updated) so leave this as false in RPI. I don't know it is solved or not.
      							actionLocale: "en-US", // At this moment, multi-languages are not supported, sorry. Someday I'll work.
    						},
    				snowboy: {
      					audioGain: 2.0,
      					Frontend: true,
      					Model: "jarvis",
      					Sensitivity: null
    					},
    				A2DServer: {
      						useA2D: false,
      						stopCommand: "stop"
    					},
    				recipes: [ "with-MMM-TelegramBot.js" ]
  					}
		},		

```

lxne, 2020-07-05:
Unfortunately this is known but some kind of a strange issue (not really solved yet).
Can you please provide a view more details.
Which Pi model, how much Memory, audio input device, audio output device?

majik655, 2020-07-05:
`cat /proc/device-tree/model`
Raspberry Pi Model (per command above)
 **Raspberry Pi 3 Model B Plus Rev 1.3** 

**Memory is 1gig per command free -m**
```
total        used        free      shared  buff/cache   available
Mem:            925         176         489          11         259         684
Swap:            99          35          64
```

Audio Output and Audio input (currently) is using a usb soundcard with speaker out output and mic input
**lsusb command shows -**
```Bus 001 Device 007: ID 1b3f:2008 Generalplus Technology Inc.```

**aplay-l command shows** 
```card 2: Device [USB Audio Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

**arecord -l command shows** 
```
**** List of CAPTURE Hardware Devices ****
card 2: Device [USB Audio Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

**The microphone I am currently using is** 
EDUTIGE ETM-001 Microphone - Omnidirectional 3.5mm 3-Pole(TRS) Microphone for GoPro, DSLR, Mirrorless Camera or Digital Audio Recorder 
[https://smile.amazon.com/gp/product/B00IP0ST78/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1](url)

And the speaker I am using is connected to the 3.5 audio output jack on the usb sound card.

Everything works command wise and listening and responding except just no sound from GA voice...
I have not tried to use HDMI for sound yet but I do not think it will matter since chimes and dings are coming through fine.

It all works (well except I havent figured out why it tries to display youtube link but doesn't get there and just goes back to main screen....but I think I see this issue from other people also.)

Let me know if I need to provide more info!
Thank you

majik655, 2020-07-05:
I want to add I just changed my swapfile to 1024 

I started MM and said to Jarvis "Thriller " and she said "this is what comes up" over the speaker! 

However I can not get her to say it again and I can't seem to get anything to really load very quickly ... it has to be ram or memory related.

bugsounet, 2020-07-05:
So... 
Issues win and take me head !

I will code dual audio support : HTML5 and with a player program
It is not my ambition at the start but i will try.
I hope A2D will still work

majik655, 2020-07-05:
@bugsounet 
I seemed to have maybe figured out the youtube playing part... I had to enable FULL KMS Open GL in raspi config.  Before I was using FAKE KMS  .  Not sure if that really had anything to do about it but now GA takes me to youtube video I haven't tried casting yet with FULL KMS but with previous FAKE KMS it wouldn't cast and would say an error has happened and wants to send a report.

I have tried a lot to get GA to respond with her voice.  It's obviously a ram thing.

majik655, 2020-07-06:
@bugsounet
Issue #74 solved no sound from GA voice for me.

Moving sound output to analog on  board 3.5 jack from usb soundcard has fixed the audio issue of both hearing GA speak and lack of sound while casting.

bugsounet, 2020-07-06:
@majik655 : it's possible because i have this config too

in all case, v2.1.0 is out with native program support for sound output

closed 2020-07-12 by bugsounet