bugsounet/MMM-GoogleAssistant #78: Help with error?

AliJDB, 2020-07-07:
Hello! I'm very new to this and generally pretty hopeless so my apologies in advance. 

I went through the installation for MMM-GoogleAssistant seemingly successfully, but now when I try to run Magic Mirror I get this recurring: 

`[2020-07-07 17:24:57.909] [LOG]    [SNOWBOY] Retry restarting...
[2020-07-07 17:24:59.977] [LOG]    [LPCM16] WARN: arecord: main:828: 
[2020-07-07 17:24:59.978] [LOG]    [LPCM16] WARN: audio open error: No such file or directory
`

Any idea what might be causing it or how to fix it? I've run out of things to try I think.

bugsounet, 2020-07-07:
this means that your MicConfig device is not properly configured

try to determinate you device with:

`arecord -l`

and report it with `device : "plughw: < nb >"`

exemple:
```sh
$ arecord -l
**** Liste des Périphériques Matériels CAPTURE ****
carte 2: Device [USB PnP Audio Device], périphérique 0: USB Audio [USB Audio]
  Sous-périphériques: 0/1
  Sous-périphérique #0: subdevice #0
```
in this example card is number 2
so result of micConfig device is `device: "plughw:2"`

AliJDB, 2020-07-07:
Thank you so much for your quick response! I got it working :) You're a superstar!

bugsounet, 2020-07-07:
Perfect! 
now let's try MMM-Assistant2Display extension ;)

closed 2020-07-12 by bugsounet