bugsounet/MMM-GoogleAssistant #79: getting error after installing assistant.

brusslesprouts, 2020-07-08:
I have the module installed and got my credentials from the google api site.   I run the command and get this response.

:~/MagicMirror/modules/MMM-GoogleAssistant $ node auth_and_test.js
internal/modules/cjs/loader.js:800
    throw err;
    ^

SyntaxError: /home/pi/MagicMirror/modules/MMM-GoogleAssistant/credentials.json: Unexpected token / in JSON at position 0
    at JSON.parse (<anonymous>)
    at Object.Module._extensions..json (internal/modules/cjs/loader.js:797:27)
    at Module.load (internal/modules/cjs/loader.js:653:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
    at Function.Module._load (internal/modules/cjs/loader.js:585:3)
    at Module.require (internal/modules/cjs/loader.js:692:17)
    at require (internal/modules/cjs/helpers.js:25:18)
    at new Auth (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/google-assistant/components/auth.js:26:15)
    at new GoogleAssistant (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/google-assistant/index.js:21:16)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/auth_and_test.js:60:19)

brusslesprouts, 2020-07-08:
now its saying i can not find my credentials.json file

brusslesprouts, 2020-07-08:
well was able to get the google link to pop up after running the node auth_and_test.js script.

but took me to a google page that said this

Authorization Error
Error 403: restricted_client
This app is not yet configured to make OAuth requests. To do that, set up the app’s OAuth consent screen in the Google Cloud Console.
Learn more
Request Details
access_type=offline
o2v=1
response_type=code
redirect_uri=urn:ietf:wg:oauth:2.0:oob
client_id=130044304074-ab875f2e9engvpep47kkv655tpcge97e.apps.googleusercontent.com
scope=https://www.googleapis.com/auth/assistant-sdk-prototype

bugsounet, 2020-07-08:
Hi, 

Generally, this error happen when you don't do the step 4 `OAuth Consent Screen setup` of the [Google Assistant Setup](https://github.com/bugsounet/MMM-GoogleAssistant/wiki/Google-Assistant-Setup)

bugsounet, 2020-07-28:
Hi, I think your problem is now solved. reopen this issue if needed

closed 2020-07-28 by bugsounet