bugsounet/MMM-GoogleAssistant #80: Assistant cannot hear me

nguyenanhminh97, 2020-07-09:
I installed MMM-GoogleAssistant successfully but the assistant didn't hear me when I called "subex". After a period of continuous retries, it was exited from the MagicMirror program

my logs
0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@6.14.4
3 info using node@v10.21.0
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle magicmirror@2.12.0~prestart: magicmirror@2.12.0
6 info lifecycle magicmirror@2.12.0~start: magicmirror@2.12.0
7 verbose lifecycle magicmirror@2.12.0~start: unsafe-perm in lifecycle true
8 verbose lifecycle magicmirror@2.12.0~start: PATH: /usr/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/home/pi/MagicMirror/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games
9 verbose lifecycle magicmirror@2.12.0~start: CWD: /home/pi/MagicMirror
10 silly lifecycle magicmirror@2.12.0~start: Args: [ '-c',
10 silly lifecycle   'DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js' ]
11 info lifecycle magicmirror@2.12.0~start: Failed to exec start script
12 verbose stack Error: magicmirror@2.12.0 start: `DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js`
12 verbose stack spawn ENOENT
12 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:48:18)
12 verbose stack     at ChildProcess.emit (events.js:198:13)
12 verbose stack     at maybeClose (internal/child_process.js:982:16)
12 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:259:5)
13 verbose pkgid magicmirror@2.12.0
14 verbose cwd /home/pi/MagicMirror
15 verbose Linux 4.19.118-v7+
16 verbose argv "/usr/bin/node" "/usr/bin/npm" "start"
17 verbose node v10.21.0
18 verbose npm  v6.14.4
19 error code ELIFECYCLE
20 error syscall spawn
21 error file sh
22 error errno ENOENT
23 error magicmirror@2.12.0 start: `DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js`
23 error spawn ENOENT
24 error Failed at the magicmirror@2.12.0 start script.
24 error This is probably not a problem with npm. There is likely additional logging output above.
25 verbose exit [ 1, true ]

bugsounet, 2020-07-09:
this log does not speak enough
can you give me the log with `npm start` please ?

nguyenanhminh97, 2020-07-09:
Thanks, it seems that I don't see the issue again, though I haven't edited it yet.
I have another question about how to check the speakers and mic again, I think GA hasn't heard me yet

nguyenanhminh97, 2020-07-10:
I fixed the problem with npm run update
thanks

closed 2020-07-10 by nguyenanhminh97