bugsounet/MMM-GoogleAssistant #81: Register your Mirror as Google Assistant related device

nguyenanhminh97, 2020-07-10:
I followed the instructions and the last step failed

(env) pi@raspberrypi:~ $ curl -s -X POST -H "Content-Type: application/json" -H "Authorization: Bearer $ACCESSTOKEN" -d @deviceInstance.json https://embeddedassistant.googleapis.com/v1alpha2/projects/YOUR_PROJECT_ID/devices/
{
  "error": {
    "code": 400,
    "message": "Request contains an invalid argument.",
    "status": "INVALID_ARGUMENT"
  }
}

deviceInstance.json
{
    "id": "MYMIRROR_001",
    "model_id": "magicmirror-b6d61",
    "nickname": "My Assistant on MAGICMIRROR",
    "client_type": "SDK_SERVICE"
  }

thanks for helping!

bugsounet, 2020-07-10:
> (env) pi@raspberrypi:~ $ curl -s -X POST -H "Content-Type: application/json" -H "Authorization: Bearer $ACCESSTOKEN" -d @deviceInstance.json https://embeddedassistant.googleapis.com/v1alpha2/projects/YOUR_PROJECT_ID/devices/

replace YOUR_MODEL_ID by your own value

bugsounet, 2020-07-28:
Hi, I think your problem is now solved. reopen this issue if needed

closed 2020-07-28 by bugsounet