bugsounet/MMM-GoogleAssistant #82: mmm-googleassistant is not working

kayoung11, 2020-07-10:
there were no error in all the installation processes, but they didn't work...
I've done config.js file modifications normally, but I can't see google assistant UI at MagicMirror.
Is there anyone else who's ever had the same issue? How did solve it? What was the problem?

kayoung11, 2020-07-10:
[image]
 
there's no error like the above capture.. Is there anything missing?

bugsounet, 2020-07-10:
Hi,

> but I can't see google assistant UI at MagicMirror

UI only appears when activated with the keyword,
Just says your prefered keyword (`Jarvis` by default)

Engkourosh, 2020-07-12:
hi i used MMM-GoogleAssitant Module but its not working auto detect when i say "jarvis" or "smart mirror" thats not response 
can you help me ?

[https://img.techpowerup.org/200712/1111.jpg](url)

Engkourosh, 2020-07-12:
this is my code  

{
			  module: "MMM-AssistantMk2",
			  position: "top_right",
			    config: {
			    verbose: true,
			    startChime: "connection.mp3",
			    useWelcomeMessage: "Hi Amir",
			    lang: "en-US",
			    coordinates: {
			      latitude: 51.5033640,
			      longitude: -0.1276250,
			    },
			    responseVoice: true,
			    responseScreen: true,
			    screenZoom: "80%",
			    screenDuration: 5000, 
			    standbyStatic: false,
			    record:  { // Full version
			      recorder: "arecord",  // Defaults to "arecord" - also supports "rec" and "sox"
			      device: "plughw:2"     // recording device (e.g.: "plughw:1")
			    },
			    play: {
			      playProgram: "mpg321",
			      playOption: [],
			    },
			    snowboy: {
			      audioGain: 2.0,
			      Frontend: false,
			      Model: "smart_mirror",
			      Sensitivity: null
			    }
			  }
		},

bugsounet, 2020-07-12:
Hi, do you read [this](https://github.com/bugsounet/MMM-AssistantMk2/wiki#assume-this) ?

I read in this page : 
**No issue and no help will be given for this version !**
**This project is considered closed**

Engkourosh, 2020-07-12:
but this is not working auto detect .........i had click on icon and this is worked

closed 2020-07-12 by bugsounet