bugsounet/MMM-GoogleAssistant #84: Google Assistant Module - No sound.

TheLaurenBarger, 2020-07-12:
Hello, 

Would you please have a look at this particular issue? I've attempted a full reinstall. I've installed this on the latest Raspian 10 Buster on a Pi 3B+ and am using the ReSpeaker 2-mic hat. Other than these modules I'm only using default magic mirror modules. I'm at a loss as to how to fix it.

There is no sound coming through the speakers. 

I've doubly made sure that the right card number is being used in the module config. I've attached logs and config below. 

Thank you so much!  

[config_and_log.zip](https://github.com/bugsounet/MMM-GoogleAssistant/files/4908376/config_and_log.zip)

bugsounet, 2020-07-12:
1) in config.js mising at the end of the file:
```
/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== "undefined") {module.exports = config;}
```
2) you have already one instence of MagicMirror launched
try to stop it:
```
pm2 stop 0
cd ~/MagicMirror
npm start
```
and give me result

lxne, 2020-07-12:
My setup is quite similar and I too had some problems with my ReSpeaker 2-mic hat.
GoogleAssistant's always plays sound with the default device. In the taskbar you can select the output device with the sound icon. But when I chose my ReSpeaker sound always got muted and the volume slider did not affect sound output at all. It seems the slider is somehow hooked to the wrong ReSpeaker function (it is affecting the 3D slider instead of volume). I tried a view things. When I removed the volume control from the taskbar and deleted .asoundrc I finally got sound output from MagicMirror. Make sure that ReSpeaker is selected as default output before removing the volume control.

TheLaurenBarger, 2020-07-13:
Hello again, and thank you for your responses!  

1. My apologies, I have the missing config file line. I didn't include it in my initial post. That line is there. 

2. Thank you! This was definitely the issue listed in the logs, I've attached the new logs and you can see that the port error is gone. 
[logs.zip](https://github.com/bugsounet/MMM-GoogleAssistant/files/4911832/logs.zip)

Sound is now coming from the mirror and that is fantastic! Thank you so much again. 

However, there is still no sound from YouTube videos? I can see there is this line appearing in the logs: 
`
ALSA lib pcm.c:8591:(snd_pcm_set_params) Unable to get period size for PLAYBACK: Invalid argument
`

I don't know if that might have something to do with it? 

@Ixne, thank you for your information! I will give your advice a try just to see if that maybe has something to do with it.

TheLaurenBarger, 2020-07-15:
Hello again, 

I see there's been an update to the module. I've pulled the changes sort of hoping that would fix the error. No luck yet though. 

Apologies, but I don't suppose you have any other ideas on what the issue could be?

artomka, 2020-07-22:
Hi, I have the same issue. I can hear the sound from mirror, but not from YouTube :(

lxne, 2020-07-22:
Did you both try 'useNative: false'?

artomka, 2020-07-22:
> Did you both try 'useNative: false'?

No, 'useNative: false' is a default option. It is a good idea to change the value to "true". I will try it today and write back. Thanx.

artomka, 2020-07-22:
Nope. It is the same. Any other suggestion?

bugsounet, 2020-07-22:
MMM-GoogleAssistant don't show any youtube video natively

what's additional module do you use ?

artomka, 2020-07-22:
> MMM-GoogleAssistant don't show any youtube video natively
> 
> what's additional module do you use ?
Your module: MMM-Assistant2Display

bugsounet, 2020-07-22:
so why posting there ?

It's not an GA issue :)
main issue is solved. GA have sound ;)
Please open an issue in the A2D module if needed

closed 2020-07-22 by bugsounet

artomka, 2020-07-22:
OK :))

artomka, 2020-07-22:
Mission completed :))