bugsounet/MMM-GoogleAssistant #85: Not Chime sound and responding please help

Engkourosh, 2020-07-12:
hi guys , i installed this module but im not see enything in my magic mirror when i say "jarvis" this is not work

i install this module according to installtion ,my google assistant work on my terminal and it response but not work on magic mirror 

please help me

Engkourosh, 2020-07-12:
this is screenshot of my terminal and my config.js file

https://img.techpowerup.org/200712/4444.jpg

https://img.techpowerup.org/200712/3333.jpg

bugsounet, 2020-07-12:
Do you verify your mic level ?

Magicmirror1991, 2020-07-12:
i tested my mic in terminal and this worked good, when i run my magic mirror i should see like this image?

https://github.com/bugsounet/MMM-GoogleAssistant/blob/dev/resources/previewFS.jpg

or when i say "jarvis" or "smart mirror" this is active and show for me?

bugsounet, 2020-07-12:
in you case, it's `jarvis` and yes it's should activate in full screen

what's the result of `arecord-l` in a terminal ?

Magicmirror1991, 2020-07-12:
but when i run my MM i never see any thing about google assistant module:(

https://img.techpowerup.org/200712/555.jpg

bugsounet, 2020-07-12:
Every seems good and ready
It seems to be a problem with mic level
in RPI desktop -> icon volume -> audio input 
 1) selec your mic
 2) audio input setting to verify your mic level (verify if it's not mutted also)

> i tested my mic in terminal and this worked good

how do you test it ?

Magicmirror1991, 2020-07-12:
its ok i checked it

https://img.techpowerup.org/200712/6666.jpg

https://img.techpowerup.org/200712/777.jpg

in during of instalation in terminal it asked me do you wanna test your mic and your speaker and i write yes and checked it work good

what is my problem?why cannt i see google assistant module in my MM?

bugsounet, 2020-07-12:
GoogleAssisant UI will only show activated when you say your keyword `Jarvis`
if you don't say it ... i will never show GA interface

Magicmirror1991, 2020-07-12:
aha ok

so what you think for my problem?i followed instructions step by step everything installed well

Magicmirror1991, 2020-07-12:
hey its worked.....but when i get my usb port and push it and i say "jarvis: its work , maybe my mic has problem?but when i used my mic on google translate it work very well :|

Magicmirror1991, 2020-07-12:
when i touch my usb port on raspberry pi its work very good but when i donnt touch that its not worked,i confused :| my mic working good in my translate google in raspberry :|

Magicmirror1991, 2020-07-12:
hey can we connected google assistant(this module) to ifttt ? what can i use for connect ifttt to raspberry ?with portforwarding or something else?

bugsounet, 2020-07-12:
GA respond to IFTTT request.
I use it with natural IFTTT commands and response
I don't use any plugin
After I Don't know what to you want to do exactly ?

Magicmirror1991, 2020-07-13:
i solved mi mic with headset bluetooth i think my usb mic has a problem

i wanna sync mt GA to iftt and with voice command controll relay on raspberry

i reggisterd on ifttt site and select the icon "then" and select the google assistant now i need to select "then" and i dont know what to do 

i should select the tp-link modem and need to setting port forwarding?for access ifttt to my raspberry pi?

need python code for that?

bugsounet, 2020-07-13:
there are a lot of methods
Personally, I ask IFTTT to send an HTTP link to the RPI and execute the command via a PHP script

Magicmirror1991, 2020-07-16:
php script ?like "particle.io" site?

any way can i ask you something, is there module for searching bluetooth device on magic mirror?for example we want to make smart mirror for end user and we donnt want user access to raspberry OS and use bluetooth option to connect any device, can we install some module to scan and connect devices(like speaker bluetooth)

bugsounet, 2020-07-16:
not to my knowledge sorry

Magicmirror1991, 2020-07-16:
there are 1 bluetooth module for Oralbe but not work for other device, do you know some body help me for this ? 

thank you

bugsounet, 2020-07-28:
Hi,
I don't know about this.
I think your main problem is now solved

closed 2020-07-28 by bugsounet