bugsounet/MMM-GoogleAssistant #88: change voice?

uniotAdmin, 2020-07-13:
1. On a smartphone, Google Assistant has two types of voices, male voice and female voice.

MMM-GoogleAssisant is a male voice.

Can I change it with a female voice?

2. It takes considerable time to understand my voice.
Is this my network speed issue?

lxne, 2020-07-13:
1. Interesting. Mine has a female voice. I didn't change anything.
2. Yes, this can be a network speed issue.

bugsounet, 2020-07-13:
Hi, for change the Assistant voice
Use the Google Home app

**I will try to explain where it is (because English is not my natural language)**
* Open the app
* top right -> your icon account
* parameter of google assistant
* Assistant tab -> voice assistant

needed the same accound for GA and the phone
and it will be set for ALL device linked

closed 2020-07-13 by wiresharkkkk