bugsounet/MMM-GoogleAssistant #91: where should i find -*configuration generated by auto-installer*-

MWHM212, 2020-07-15:
hi again 😅
i need to edit Configuration file "config.js"

in simple section i find this line 

    micConfig: { // put there configuration generated by auto-installer

i didnt understand where should i find    configuration generated by auto-installer  ??

help me ASAP pls 
thx

bugsounet, 2020-07-15:
it's last part of `npm install`
```
Checking Speaker and Microphone...
Do you want check your audio output configuration ?
```
you can again run this script with `ǹpm run update`

MWHM212, 2020-07-15:
Did you hear yourself? [Y/n]
Your choice: y
This is your GoogleAssistant micConfig working configuration :

micConfig: {
recorder: "arecord",
device: "plughw:1"
},

MMM-GoogleAssistant is now installed !
Press [Enter] to continue

its empty ..

MWHM212, 2020-07-15:
maybe there is something missing in SNOWBOY file 

[2020-07-15 15:40:57.124] [LOG] [SNOWBOY] Retry restarting... 
[2020-07-15 15:40:59.212] [LOG] [LPCM16] WARN: ALSA lib pcm_hw.c:1822:(_snd_pcm_hw_open) Invalid value for card arecord: main:828: 
[2020-07-15 15:40:59.213] [LOG] [LPCM16] WARN: audio open error: No such file or directory

bugsounet, 2020-07-15:
and what's the result of:
`arecord -l`

MWHM212, 2020-07-15:
pi@raspberrypi:~/MagicMirror $ arecord -l
**** List of CAPTURE Hardware Devices ****
card 1: Device [USB Audio Device], device 0: USB Audio [USB Audio]
   Subdevices: 1/1
   Subdevice #0: subdevice #0

is that what you need to know ??

bugsounet, 2020-07-15:
and what is your `micConfig : {}` in GA config ?

MWHM212, 2020-07-15:
pi@raspberrypi:~/MagicMirror/modules/MMM-GoogleAssistant $ ls

auth_and_test.js
 MMM-GoogleAssistant.js
 resources
components 
node_helper.js
 screen_output.css
credentials.json
 node_modules 
setup
deviceInstance.sample.json
 package.json 
tmp
installer
 package-lock.json
 token.json
LICENSE
 README.md 
translations
MMM-GoogleAssistant.css 
recipes

there is no config file in here !?

bugsounet, 2020-07-15:
MagicMirror `config.js` file can help

MWHM212, 2020-07-15:
/* Magic Mirror Config Sample
 *
 * By Michael Teeuw http://michaelteeuw.nl
 * MIT Licensed.
 *
 * For more information on how you can configure this file
 * See https://github.com/MichMich/MagicMirror#configuration
 *
 */

var config = {
	address: "localhost", // Address to listen on, can be:
	                      // - "localhost", "127.0.0.1", "::1" to listen on loopback interface
	                      // - another specific IPv4/6 to listen on a specific interface
	                      // - "0.0.0.0", "::" to listen on any interface
	                      // Default, when address config is left out or empty, is "localhost"
	port: 8080,
	ipWhitelist: ["127.0.0.1", "::ffff:127.0.0.1", "::1"], // Set [] to allow all IP addresses
	                                                       // or add a specific IPv4 of 192.168.1.5 :
	                                                       // ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.1.5"],
	                                                       // or IPv4 range of 192.168.3.0 --> 192.168.3.15 use CIDR format :
	                                                       // ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.3.0/28"],

	useHttps: false, 		// Support HTTPS or not, default "false" will use HTTP
	httpsPrivateKey: "", 	// HTTPS private key path, only require when useHttps is true
	httpsCertificate: "", 	// HTTPS Certificate path, only require when useHttps is true

	language: "en",
	timeFormat: 24,
	units: "metric",
	// serverOnly:  true/false/"local" ,
			     // local for armv6l processors, default
			     //   starts serveronly and then starts chrome browser
			     // false, default for all  NON-armv6l devices
			     // true, force serveronly mode, because you want to.. no UI on this device

	modules: [
		{
			module: "alert",
		},
		{
			module: "updatenotification",
			position: "top_bar"
		},
		{
			module: "clock",
			position: "top_left"
		},
		{
			module: "calendar",
			header: "US Holidays",
			position: "top_left",
			config: {
				calendars: [
					{
						symbol: "calendar-check",
						url: "webcal://www.calendarlabs.com/ical-calendar/ics/76/US_Holidays.ics"					}
				]
			}
		},
		{
			module: "compliments",
			position: "lower_third"
		},
		{
		  module: "MMM-GoogleAssistant",
		  position: "fullscreen_above",
		  config: {
		    assistantConfig: {
		      latitude: 51.508530,
		      longitude: -0.076132,
		    },
		    micConfig: {
		      recorder: "arecord",
		      device: "plughw:2",
		    },
		  }
		},
		{
			module: "currentweather",
			position: "top_right",
			config: {
				location: "New York",
				locationID: "", //ID from http://bulk.openweathermap.org/sample/city.list.json.gz; unzip the gz file and find your city
				appid: "YOUR_OPENWEATHER_API_KEY"
			}
		},
		{
			module: "weatherforecast",
			position: "top_right",
			header: "Weather Forecast",
			config: {
				location: "New York",
				locationID: "5128581", //ID from http://bulk.openweathermap.org/sample/city.list.json.gz; unzip the gz file and find your city
				appid: "YOUR_OPENWEATHER_API_KEY"
			}
		},
		{
			module: "newsfeed",
			position: "bottom_bar",
			config: {
				feeds: [
					{
						title: "New York Times",
						url: "http://www.nytimes.com/services/xml/rss/nyt/HomePage.xml"
					}
				],
				showSourceTitle: true,
				showPublishDate: true,
				broadcastNewsFeeds: true,
				broadcastNewsUpdates: true
			}
		},
	]

};

/************* DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== "undefined") {module.exports = config;}

bugsounet, 2020-07-15:
so ... take place to your device config

MWHM212, 2020-07-15:
thx you a lot 
i think that i dont have problems but the GA doesn't hear me when i speak
and there is no GA sign in MM ..

bugsounet, 2020-07-15:
Verify your mic level in RPI desktop
 -> icon volume -> audio input

selec your mic
audio input setting to verify your mic level (verify if it's not mutted also)

where do you read there is an GA sign in MM??
It's not needed and never i will make it ;)

MWHM212, 2020-07-15:
you are the best man  ;)

i test my mic on https://mictests.com/ and it work
then i play a youtube video it work

but when i say hey google or okay google it still the same..

bugsounet, 2020-07-15:
wow ... where do you read `hey google` or `okay google` is the keyword for activate assistant ?
Maybe, it's time for you to read the [wiki](https://github.com/bugsounet/MMM-GoogleAssistant/wiki)  pages.

MWHM212, 2020-07-15:
i searched a lot but i didnt find the keyword😅😢
what is the keyword or how could i change it ??.

bugsounet, 2020-07-15:
> Maybe, it's time for you to read the wiki pages.

Sorry, but i can't explain all possibilities of [configuration](https://github.com/bugsounet/MMM-GoogleAssistant/wiki/Configuration-Sample)
If you don't want to read, i can't help you more.
If it's too complex for you, just don't use this module ;)

Notes:
* An issue is a problem with the main program
 --> this issue is related to this ? no ...
* This issue can't resolve humain problem
 --> I'm ready to help but there are limits however
 --> If you don't make an effort, I won't do my part
 --> issue is not an assistance chat
 --> your reactions don't make me laugh

So, your main in title issue is now solved, perfect
You can open another issue, but you will have no helping

closed 2020-07-15 by bugsounet

MWHM212, 2020-07-15:
no you just understand me wrongly 
i am ready to do anything to show effort but i didnt know what should i read 

anyway thank you for all attention that you give to me :)
and i really I appreciate your work , keep going ..