bugsounet/MMM-GoogleAssistant #92: Google Home Issue after registering Mirror/Assistant

Metalopn, 2020-07-16:
Hey, firstly I love your work. Secondly, I've got a problem. I managed to get all the way through all of the setup, install, registration and everything. Once I did the device registration so I could see the mirror/assistant on google home, it showed up but when I click on the mirror is only shows it as "offline" and I can't use it for anything on the app. Since everything seemed to go well, do you have any idea where I may have gone wrong? Is this normal behavior?

wirdman03, 2020-07-17:
can you help me i got an error when i run 

`pi@raspberrypi:~ $ curl -s -X POST -H "Content-Type: application/json" -H "Authorization: Bearer $ya29.a0AfH6SMDVmMj1ceMvjYTS6aahSpK8jZgwRlNKoG-VI9oyo2yoBvJJP-t25Xle7iAlaHb7f3EqecSSTZ0UfacKrMOeZtw0pOnNxbc2WV2Tt5xtzzehNLg2FJvreaMjDaiVXT2YGoqptkDwdMRvOTcwOd9w80_6q94vJ94" -d @deviceInstance.json https://embeddedassistant.googleapis.com/v1alpha2/projects/13_06_2020/devices/x`

error 

`curl -s -X POST -H "Content-Type: application/json" -H "Authorization: Bearer $ya29.a0AfH6SMDVmMj1ceMvjYTS6aahSpK8jZgwRlNKoG-VI9oyo2yoBvJJP-t25Xle7iAlaHb7f3EqecSSTZ0UfacKrMOeZtw0pOnNxbc2WV2Tt5xtzzehNLg2FJvreaMjDaiVXT2YGoqptkDwdMRvOTcwOd9w80_6q94vJ94" -d @deviceInstance.json https://embeddedassistant.googleapis.com/v1alpha2/projects/13_06_2020/devices/`

Metalopn, 2020-07-18:
Try removing the "$" from infront of the authorization code:

Is:
Bearer $ya29.a0

Try:
Bearer ya29.a0

phungorquan, 2020-07-18:
Hi, could you help me about credentials of google home.
Follow this module to setup, we have to download the credentials,.... . But my friend are researching about google home with fullfillment system. I want to integrate it to my G.A but i could not find out the credentials on action console site :(.
Thanks!

bugsounet, 2020-07-28:
MMM-GoogleAssistant is a SDK project (+/- an Google Assisant Demo) so your feature is not inclued with a Google Home
maybe by gactions but i can't help you about this. (I never try to do this)

closed 2020-07-28 by bugsounet