bugsounet/MMM-GoogleAssistant #93: Invoke recipe

phungorquan, 2020-07-16:
Hi,
I tried to and some recipe in 'config.js' as sample config but i can not invoke(call) pattern. I tried "reboot please" and it search on google. The G.A did not search in recipe. 
How can i avoid this? :D

Thanks

bugsounet, 2020-07-16:
1) maybe there is a callback sended by Google and it's block the recipe (priority order: google --> actions -> recipe)
if google have a callback -> recipe will not bee executed
if gaction callback  -> recipe will not be executed too

try with another pattern without reboot word

2) the volume control is Asistant2Display job

phungorquan, 2020-07-17:
> 1. maybe there is a callback sended by Google and it's block the recipe (priority order: google --> actions -> recipe)
>    if google have a callback -> recipe will not bee executed
>    if gaction callback  -> recipe will not be executed too
> 
> try with another pattern without reboot word
> 
> 1. the volume control is Asistant2Display job

But i think google has everything :(, it's hard to say something without be searching from google .

I don't understand what you mean about A2D, because i saw the sound system on rasp did not change :(

bugsounet, 2020-07-17:
natively, I don't code volume control in MMM-GoogleAssistant (GA)
Google return a callback and GA send it to MMM-Assistant2Display (A2D)
so only  A2D (with volume sub-module) have this fonctionality

Note: 
 * google voice return don't do natively his job without coding of GA or A2D for the google callback ;)
 * google send audio and I just playing it (I can't control it)
 * for audio control: Google don't know what is your RPI volume and so google server believe volume is 100%
 * you can send an order to force volume at 50% ... and it do a callback (with gaction)
 * google server don't remember volume value set and reset it to 100% only RPI have the good value
 
Just remember this:
This project use a SDK of google assistant, it +/- of a demo of google assisant
I just try to exploit of google callback (when it's sended because it's not always the case)

I hope I answered your questions ;)
sorry for my english because it's not my natural language ;)

phungorquan, 2020-07-18:
Hi, i can invoke my recipe but i think i should find some private words for me :D.

About sound, i was not clear what you say but you mean i have to setup action to execute the command in my recipe, which to control volume. Right?

bugsounet, 2020-07-18:
just increase you volume mic -> RPI destop -> icon volume -> (right click) -> output device -> setting

for volume control, If you use [A2D](https://github.com/bugsounet/MMM-Assistant2Display), no recipe is needed, just turn on sub-module of it. You can see A2D wiki for more information

phungorquan, 2020-07-18:
Sorry for missing, I read the A2D configuration and now i can control volume via voice!!!

This is my 'config.js' for who don't know:
`volume: {
      useVolume: true,
      volumePreset: null ,
      myScript: "amixer sset Headphone #VOLUME#%"
},`
To know why i know Headphone, use command "amixer" , and enjoy :D

THANKSSSSS !

bugsounet, 2020-07-18:
sure ;)
It's just equal to `volumePreset: "ALSA_HEADPHONE"` [wiki](https://github.com/bugsounet/MMM-Assistant2Display/wiki/volume) can help sometime :)

closed 2020-07-21 by bugsounet