bugsounet/MMM-GoogleAssistant #95: MM stopped working after latest changes/updates from GA

arunfruity, 2020-07-17:
Hi There,

I got a notification that GA has new update and hence i pulled the latest from GIT and installed. Now when I start up I get this error message 

WARNING! Could not load config file. Starting with default configuration. Error found: Error: Cannot find module '@bugsounet/md5'
Require stack:
- /home/pi/MagicMirror/modules/MMM-Assistant2Display/node_helper.js
- /home/pi/MagicMirror/js/app.js
- /home/pi/MagicMirror/js/electron.js
- /home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js

bugsounet, 2020-07-17:
run the natural command to update A2D and GA on each directory
```
npm run update
```
and it will works ;)

arunfruity, 2020-07-17:
That's worked thanks

closed 2020-07-17 by arunfruity