bugsounet/MMM-GoogleAssistant #96: Interrupt Google Assistant response and Scroll website

phungorquan, 2020-07-18:
Hi,
I want to ask, Can we interrupt google assistant response? 
Because when i ask G.A to search something, during it's reading the content i could not say 'Jarvis' or something to search the new things until it finish.

By the way, without auto scrolling when search on wiki or website, can we control scroll via mouse? I saw we can not interact, or click into the screen.

Thanks! :D

bugsounet, 2020-07-18:
> Can we interrupt google assistant response?

actually not yet coded

> without auto scrolling when search on wiki or website, can we control scroll via mouse? I saw we can not interact, or click into the screen.

yes with a mouse you can (i think)
you can stop the process with your prefered stop command (see [wiki there](https://github.com/bugsounet/MMM-GoogleAssistant/wiki/Configuration-Structure#field-a2dserver))

phungorquan, 2020-07-18:
I hope you will develop to interrupt G.A :D. 

About volume control, sorry for missing, I read the A2D configuration and now i can control volume via voice!!!

This is my 'config.js' for who don't know:
`volume: {
      useVolume: true,
      volumePreset: null ,
      myScript: "amixer sset Headphone #VOLUME#%"
},`

To know why i know Headphone, use command "amixer" , and enjoy :D

closed 2020-07-28 by bugsounet