bugsounet/MMM-GoogleAssistant #101: GA is not working

nguyenanhminh97, 2020-07-22:
when I call jarvis, my GA is activated and receives my command (for example: what time is it), and then GA does not work, does not exit the command prompt, the GA mic icon is running continuously continuous and non-stop.
I can only fix it with ctrl + q and restart GA
I want to ask if there is any way to fix the error or the modules / features that help me reload MM automatically in the above case (replace ctrl + q and restart MM)
thanks

lxne, 2020-07-23:
If it is a GUI crash you could try https://github.com/MichMich/MMM-WatchDog

For more help, more information would be needed.
Hardware, MagicMirror version, other modules used ...

bugsounet, 2020-08-01:
> the GA mic icon is running continuously continuous and non-stop

Euh... There no mic icon in this module. It's Only on GA v1
In latest version there is not this. 
Please update the module with a clean install. And read wiki

closed 2020-08-01 by bugsounet