bugsounet/MMM-GoogleAssistant #103: can i get help on MMM-AssistantMk2 pls

MWHM212, 2020-07-22:
hmm.. 
i know that i am toooo Annoying you but pls help me ASAP because i need to finish it as earlier i can 
i am so exited to see the project done 
pls dont see me as a bad person 
i love mbappe too 😅😅

bugsounet, 2020-07-23:
I close AMk2 issue system because I don't help you an outdate assistant system

If you want to use this new project i will give help (modern Google Assitant system)
so make you choice ... you can see a demo video [there](https://youtu.be/viE9Ds8IvUQ) and you will understand why if you see entire video...

MWHM212, 2020-07-23:
my entire attention is to speak with GA and she speak with me too like i need to hear the GA , i tried the MMM-GoogleAssistant module but the GA does not speaking  , that all what i need 

if this module that you showed to me is speaking i will do it

bugsounet, 2020-07-24:
Ok ... you want help for this very older module ... i wil give you this

* I have to know, your AMk2 config ?
* can you give me the result of the terminal `arecord -l` command ?
* can you give me the result of `aplay -l` command ?
* Do you set your prefered output device ?
 ---> in the RPI Desktop -> volume icon -> output device -> select your device
 ---> in the same scheme verify if not muted and verify volume
* In the same scheme too, select and verify your input device (mic)
* when all is set done: shutdown and reboot your RPI
* restart your MagicMirror app

MWHM212, 2020-07-24:
1
              },
		{
		  module: "MMM-AssistantMk2",
		  position: "top_right",
		  config: {
		    debug: false,
		    coordinates: {
		      latitude: 51.5033640,
		      longitude: -0.1276250,
		    },
		    record: { // put there configuration generated by auto-installer
		      recorder : "arecord",
		      device: "plughw:1"
		    },
		  }
		},
		{
2
pi@raspberrypi:~ $ arecord -l
** List of CAPTURE Hardware Devices **
card 1: Device [USB Audio Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
3 
pi@raspberrypi:~ $ aplay -l
** List of PLAYBACK Hardware Devices **
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 7/7
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 IEC958/HDMI [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: Device [USB Audio Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

4 yes

MWHM212, 2020-07-25:
i reinstalled the module and i started from the beginning but still there is no mic icon change

bugsounet, 2020-07-25:
i think, you must have some other program that uses the mic. can you paste me your complete configuration?

MWHM212, 2020-07-28:
/* Magic Mirror Config Sample
 *
 * By Michael Teeuw http://michaelteeuw.nl
 * MIT Licensed.
 *
 * For more information on how you can configure this file
 * See https://github.com/MichMich/MagicMirror#configuration
 *
 */

var config = {
	address: "localhost", // Address to listen on, can be:
	                      // - "localhost", "127.0.0.1", "::1" to listen on loopback interface
	                      // - another specific IPv4/6 to listen on a specific interface
	                      // - "0.0.0.0", "::" to listen on any interface
	                      // Default, when address config is left out or empty, is "localhost"
	port: 8080,
	ipWhitelist: ["127.0.0.1", "::ffff:127.0.0.1", "::1"], // Set [] to allow all IP addresses
	                                                       // or add a specific IPv4 of 192.168.1.5 :
	                                                       // ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.1.5"],
	                                                       // or IPv4 range of 192.168.3.0 --> 192.168.3.15 use CIDR format :
	                                                       // ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.3.0/28"],

	useHttps: false, 		// Support HTTPS or not, default "false" will use HTTP
	httpsPrivateKey: "", 	// HTTPS private key path, only require when useHttps is true
	httpsCertificate: "", 	// HTTPS Certificate path, only require when useHttps is true

	language: "en",
	timeFormat: 24,
	units: "metric",
	// serverOnly:  true/false/"local" ,
			     // local for armv6l processors, default
			     //   starts serveronly and then starts chrome browser
			     // false, default for all  NON-armv6l devices
			     // true, force serveronly mode, because you want to.. no UI on this device

	modules: [
		{
			module: "alert",
		},
		{
			module: "updatenotification",
			position: "top_bar"
		},
		{
			module: "clock",
			position: "top_left"
		},
		{
			module: "calendar",
			header: "US Holidays",
			position: "top_left",
			config: {
				calendars: [
					{
						symbol: "calendar-check",
						url: "webcal://www.calendarlabs.com/ical-calendar/ics/76/US_Holidays.ics"					}
				]
			}
		},
		{
			module: "compliments",
			position: "lower_third"
		},
		{

		  module: "MMM-AssistantMk2",

		  position: "top_right",

		  config: {

		    debug: false,

		    coordinates: {

		      latitude: 51.5033640,

		      longitude: -0.1276250,

		    },

		    record: { // put there configuration generated by auto-installer

		      recorder : "arecord",

		      device: "plughw:1"

		    },

		  }

		},
		{
			module: "currentweather",
			position: "top_right",
			config: {
				location: "New York",
				locationID: "", //ID from http://bulk.openweathermap.org/sample/city.list.json.gz; unzip the gz file and find your city
				appid: "YOUR_OPENWEATHER_API_KEY"
			}
		},
		{
			module: "weatherforecast",
			position: "top_right",
			header: "Weather Forecast",
			config: {
				location: "New York",
				locationID: "5128581", //ID from http://bulk.openweathermap.org/sample/city.list.json.gz; unzip the gz file and find your city
				appid: "YOUR_OPENWEATHER_API_KEY"
			}
		},
		{
			module: "newsfeed",
			position: "bottom_bar",
			config: {
				feeds: [
					{
						title: "New York Times",
						url: "http://www.nytimes.com/services/xml/rss/nyt/HomePage.xml"
					}
				],
				showSourceTitle: true,
				showPublishDate: true,
				broadcastNewsFeeds: true,
				broadcastNewsUpdates: true
			}
		},
	]

};

/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== "undefined") {module.exports = config;}

bugsounet, 2020-07-28:
and what `debug` mode says ?

MWHM212, 2020-07-28:
in which file ??

bugsounet, 2020-07-28:
set `debug: true` in AMk2 config
if you use pm2: stop the process and run it manually with:
```
cd ~/MagicMirror
npm start
```

MWHM212, 2020-07-28:
[2020-07-28 13:38:50.819] [LOG]    [SNOWBOY] Stops listening.
[2020-07-28 13:38:50.819] [LOG]    [SNOWBOY] Retry restarting...
[2020-07-28 13:38:50.820] [LOG]    [LPCM16] Stop listening
[2020-07-28 13:38:52.853] [LOG]    [SNOWBOY] Starts listening.
[2020-07-28 13:38:52.914] [LOG]    [LPCM16] ~v1.0.0~ Start listening 1 channels, use arecord with sample rate 16000
[2020-07-28 13:38:52.940] [LOG]    [LPCM16] WARN: arecord: main:828: 
[2020-07-28 13:38:52.943] [LOG]    [LPCM16] WARN: audio open error: Device or resource busy

what is that mean

bugsounet, 2020-07-28:
It's mean you already use the mic with another program

MWHM212, 2020-07-28:
i closed that progrram and start i again but stil no mic icon

closed 2020-09-08 by bugsounet