bugsounet/MMM-GoogleAssistant #104: node auth_and_test.js  command gives this error

NilanjPatel, 2020-07-23:
`pi@raspberrypi:~/MagicMirror/modules/MMM-GoogleAssistant $ node auth_and_test.js 
/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/grpc/src/grpc_extension.js:58
    throw e;
    ^

Error: Failed to load /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/grpc/src/node/extension_binary/node-v64-linux-arm-glibc/grpc_node.node. /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/grpc/src/node/extension_binary/node-v64-linux-arm-glibc/grpc_node.node: Unable to run arch-specific checks

    at Object.Module._extensions..node (internal/modules/cjs/loader.js:807:18)
    at Module.load (internal/modules/cjs/loader.js:653:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
    at Function.Module._load (internal/modules/cjs/loader.js:585:3)
    at Module.require (internal/modules/cjs/loader.js:692:17)
    at require (internal/modules/cjs/helpers.js:25:18)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/grpc/src/grpc_extension.js:32:13)
    at Module._compile (internal/modules/cjs/loader.js:778:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
    at Module.load (internal/modules/cjs/loader.js:653:32)
`

I tried to rebuild npm and also reinstall , plus I also tried npm install --target_arch=armv7 but still giving same error

bugsounet, 2020-07-23:
hi, wiki says : **`No other npm commands are needed after installation (it can broken dependencies package)`**
so retry with : `npm run rebuild` command and don't use other npm command after script done and retry
`node auth_and_test.js` process

bugsounet, 2020-08-01:
Module updated. This library is now not use.
Retry with updated version

closed 2020-08-01 by bugsounet