bugsounet/MMM-GoogleAssistant #105: Mirror crashes on hotword

JPGlunz, 2020-07-24:
The log is as follows, any help is appreciated:

```
0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@6.10.3
3 info using node@v10.22.0
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle magicmirror@2.12.0~prestart: magicmirror@2.12.0
6 info lifecycle magicmirror@2.12.0~start: magicmirror@2.12.0
7 verbose lifecycle magicmirror@2.12.0~start: unsafe-perm in lifecycle true
8 verbose lifecycle magicmirror@2.12.0~start: PATH: /usr/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/home/pi/MagicMirror/node_modules/.bin:/home/pi/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games
9 verbose lifecycle magicmirror@2.12.0~start: CWD: /home/pi/MagicMirror
10 silly lifecycle magicmirror@2.12.0~start: Args: [ '-c',
10 silly lifecycle   'DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js' ]
11 info lifecycle magicmirror@2.12.0~start: Failed to exec start script
12 verbose stack Error: magicmirror@2.12.0 start: `DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js`
12 verbose stack spawn ENOENT
12 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:48:18)
12 verbose stack     at ChildProcess.emit (events.js:198:13)
12 verbose stack     at maybeClose (internal/child_process.js:982:16)
12 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:259:5)
13 verbose pkgid magicmirror@2.12.0
14 verbose cwd /home/pi/MagicMirror
15 verbose Linux 5.4.51-v7l+
16 verbose argv "/usr/bin/node" "/usr/bin/npm" "start"
17 verbose node v10.22.0
18 verbose npm  v6.10.3
19 error code ELIFECYCLE
20 error syscall spawn
21 error file sh
22 error errno ENOENT
23 error magicmirror@2.12.0 start: `DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js`
23 error spawn ENOENT
24 error Failed at the magicmirror@2.12.0 start script.
24 error This is probably not a problem with npm. There is likely additional logging output above.
25 verbose exit [ 1, true ]
```

bugsounet, 2020-07-24:
Hi, this log don't speak enough.

Can you past the output console when you use `npm start` command ?
note: don't forget to stop pm2 if you use it

JPGlunz, 2020-07-24:
This is the output after using `npm start` and having the program crash after saying "Jarvis".

```
> magicmirror@2.12.0 start /home/pi/MagicMirror
> DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js

[2020-07-24 09:39:59.985] [LOG]    Starting MagicMirror: v2.12.0
[2020-07-24 09:39:59.992] [LOG]    Loading config ...
[2020-07-24 09:39:59.997] [LOG]    Loading module helpers ...
[2020-07-24 09:40:00.883] [LOG]    Initializing new module helper ...
[2020-07-24 09:40:00.884] [LOG]    Module helper loaded: MMM-GoogleAssistant
[2020-07-24 09:40:00.884] [LOG]    No helper found for module: MMM-MagicMover.
[2020-07-24 09:40:00.885] [LOG]    No helper found for module: clock.
[2020-07-24 09:40:01.154] [LOG]    Initializing new module helper ...
[2020-07-24 09:40:01.154] [LOG]    Module helper loaded: MMM-Reddit
[2020-07-24 09:40:01.155] [LOG]    No helper found for module: currentweather.
[2020-07-24 09:40:01.155] [LOG]    No helper found for module: weatherforecast.
[2020-07-24 09:40:01.157] [LOG]    Initializing new module helper ...
[2020-07-24 09:40:01.158] [LOG]    Module helper loaded: MMM-COVID19
[2020-07-24 09:40:01.159] [LOG]    Initializing new module helper ...
[2020-07-24 09:40:01.160] [LOG]    Module helper loaded: MMM-cryptocurrency
[2020-07-24 09:40:01.630] [LOG]    Initializing new module helper ...
[2020-07-24 09:40:01.631] [LOG]    Check MagicMirror version for node helper 'MMM-Strava' - Minimum version:  2.2.0 - Current version: 2.12.0
[2020-07-24 09:40:01.632] [LOG]    Version is ok!
[2020-07-24 09:40:01.632] [LOG]    Module helper loaded: MMM-Strava
[2020-07-24 09:40:01.633] [LOG]    No helper found for module: MMM-EasyPix.
[2020-07-24 09:40:01.633] [LOG]    No helper found for module: MMM-wiki.
[2020-07-24 09:40:01.635] [LOG]    Initializing new module helper ...
[2020-07-24 09:40:01.635] [LOG]    Module helper loaded: MMM-MWWordOfTheDay
[2020-07-24 09:40:01.986] [LOG]    Initializing new module helper ...
[2020-07-24 09:40:01.987] [LOG]    Module helper loaded: MMM-NowPlayingOnSpotify
[2020-07-24 09:40:01.989] [LOG]    Initializing new module helper ...
[2020-07-24 09:40:01.990] [LOG]    Module helper loaded: MMM-ip
[2020-07-24 09:40:02.002] [LOG]    Initializing new module helper ...
[2020-07-24 09:40:02.003] [LOG]    Module helper loaded: MMM-SystemStats
[2020-07-24 09:40:02.101] [LOG]    Initializing new module helper ...
[2020-07-24 09:40:02.101] [LOG]    Module helper loaded: MMM-NetworkConnection
[2020-07-24 09:40:02.210] [LOG]    Initializing new module helper ...
[2020-07-24 09:40:02.210] [LOG]    Module helper loaded: MMM-Remote-Control
[2020-07-24 09:40:02.211] [LOG]    All module helpers loaded.
[2020-07-24 09:40:02.312] [LOG]    Starting server on port 8080 ... 
[2020-07-24 09:40:02.322] [LOG]    Server started ...
[2020-07-24 09:40:02.323] [LOG]    Connecting socket for: MMM-GoogleAssistant
[2020-07-24 09:40:02.324] [LOG]    Connecting socket for: MMM-Reddit
[2020-07-24 09:40:02.325] [LOG]    Starting module helper: MMM-Reddit
[2020-07-24 09:40:02.325] [LOG]    Connecting socket for: MMM-COVID19
[2020-07-24 09:40:02.326] [LOG]    Starting node helper for: MMM-COVID19
[2020-07-24 09:40:02.327] [LOG]    Connecting socket for: MMM-cryptocurrency
[2020-07-24 09:40:02.327] [LOG]    Cryptocurrency module loaded!
[2020-07-24 09:40:02.328] [LOG]    Connecting socket for: MMM-Strava
[2020-07-24 09:40:02.329] [LOG]    Starting module helper: MMM-Strava
[2020-07-24 09:40:02.330] [LOG]    Connecting socket for: MMM-MWWordOfTheDay
[2020-07-24 09:40:02.331] [LOG]    Starting module helper: MMM-MWWordOfTheDay
[2020-07-24 09:40:02.332] [LOG]    Connecting socket for: MMM-NowPlayingOnSpotify
[2020-07-24 09:40:02.333] [LOG]    Connecting socket for: MMM-ip
[2020-07-24 09:40:02.334] [LOG]    Starting module helper: MMM-ip
[2020-07-24 09:40:02.334] [LOG]    Connecting socket for: MMM-SystemStats
[2020-07-24 09:40:02.335] [LOG]    Connecting socket for: MMM-NetworkConnection
[2020-07-24 09:40:02.335] [LOG]    MMM-NetworkConnection helper started ...
[2020-07-24 09:40:02.335] [LOG]    Connecting socket for: MMM-Remote-Control
[2020-07-24 09:40:02.336] [LOG]    Starting node helper for: MMM-Remote-Control
[2020-07-24 09:40:02.342] [LOG]    Sockets connected & modules started ...
[2020-07-24 09:40:02.438] [LOG]    Launching application.
[2020-07-24 09:40:06.182] [ERROR]  (node:2758) [DEP0005] DeprecationWarning: Buffer() is deprecated due to security and usability issues. Please use the Buffer.alloc(), Buffer.allocUnsafe(), or Buffer.from() methods instead.
[2020-07-24 09:40:06.184] [ERROR]  (node:2758) [DEP0005] DeprecationWarning: Buffer() is deprecated due to security and usability issues. Please use the Buffer.alloc(), Buffer.allocUnsafe(), or Buffer.from() methods instead.
[2020-07-24 09:40:06.475] [LOG]    Refreshed access token because it has expired. Expired at: 09:40:06 now is: 09:40:06
[2020-07-24 09:40:06.537] [LOG]    [ASSISTANT] MMM-GoogleAssistant Version: 2.3.1
[2020-07-24 09:40:06.780] [LOG]    [SNOWBOY] Model selected: jarvis
[2020-07-24 09:40:06.785] [LOG]    [SNOWBOY] snowboy v1.5.0 Initialized...
[2020-07-24 09:40:06.788] [LOG]    [ASSISTANT] Google Assistant is initialized.
[2020-07-24 09:40:08.093] [LOG]    starting network connection testing
/home/pi/MagicMirror/node_modules/electron/dist/electron js/electron.js: symbol lookup error: /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/grpc/src/node/extension_binary/electron-v6.1-linux-arm-gl [... cut ...]
npm ERR! code ELIFECYCLE
npm ERR! syscall spawn
npm ERR! file sh
npm ERR! errno ENOENT
npm ERR! magicmirror@2.12.0 start: `DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js`
npm ERR! spawn ENOENT
npm ERR! 
npm ERR! Failed at the magicmirror@2.12.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2020-07-24T13_40_22_546Z-debug.log

```

bugsounet, 2020-07-24:
ok, thx ;)

This error happen when you use other npm command after installation... (like `npm fix`, `npm rebuild`, `npm audix fix`, etc)
[in the wiki installation](https://github.com/bugsounet/MMM-GoogleAssistant/wiki/Installation#auto-install-for-raspberrypi-or-debian-linux-machines-only) page, you can see this :
`No other npm commands are needed after installation (it can broken dependencies package) !`

So: 
retry entire process with `npm run rebuild` command.

JPGlunz, 2020-07-24:
That did the trick! I saw that on the wiki, but for some reason I would run `npm audit fix` anyways haha. Thanks for the help!

closed 2020-07-24 by JPGlunz